davidusb-geek/emhass #401: Can not choose 'mlforecast' in Load forecast method with the web GUI

overas, 2024-12-25:
``Emhass addon v0.11.2 
Hassos 2024.10.4

When I try to select mlforecast in the web GUI there is no option for it. The choices I have are naive and csv. Is this correct?
[image]

If I manually edit the config.json file to use mlforecast and run a daily optimization everything works.
[image]

```2024-11-03 15:41:28,133 - web_server - INFO - Passed runtime parameters: {'load_cost_forecast': [1.018, 1.025, 1.027, 1.015, 1.009, 1.01, 1.007, 0.98, 1.007, 1.01, 1.01, 1.014, 1.014, 1.024, 1.052, 1.268, 1.529, 1.525, 1.498, 1.318, 1.239, 1.218, 1.202, 1.211], 'prod_price_forecast': [0.61, 0.618, 0.62, 0.607, 0.601, 0.603, 0.6, 0.572, 0.599, 0.603, 0.603, 0.606, 0.607, 0.617, 0.644, 0.741, 2.141, 2.102, 1.829, 0.79, 0.711, 0.69, 0.675, 0.683], 'prediction_horizon': 24, 'delta_forecast': 2}
2024-11-03 15:41:28,133 - web_server - INFO -  >> Setting input data dict
2024-11-03 15:41:28,133 - web_server - INFO - Setting up needed data
2024-11-03 15:41:28,135 - web_server - INFO - Retrieving weather forecast data using method = scrapper
2024-11-03 15:41:29,485 - web_server - INFO - Retrieving data from hass for load forecast using method = mlforecaster
2024-11-03 15:41:29,486 - web_server - INFO - Retrieve hass get data method initiated...
2024-11-03 15:41:31,416 - web_server - DEBUG - Number of ML predict forcast data generated (lags_opt): 60
2024-11-03 15:41:31,416 - web_server - DEBUG - Number of forcast dates obtained: 24
2024-11-03 15:41:31,418 - web_server - INFO -  >> Performing dayahead optimization...
2024-11-03 15:41:31,419 - web_server - INFO - Performing day-ahead forecast optimization
2024-11-03 15:41:31,430 - web_server - INFO - Perform optimization for the day-ahead
2024-11-03 15:41:31,576 - web_server - DEBUG - Deferrable load 0: Proposed optimization window: 0 --> 0
2024-11-03 15:41:31,577 - web_server - DEBUG - Deferrable load 0: Validated optimization window: 0 --> 0
2024-11-03 15:41:31,578 - web_server - WARNING - Solver default unknown, using default
Welcome to the CBC MILP Solver ```

davidusb-geek, 2024-12-25:
@GeoDerp I tried to look at this but got a bit lost in this js cripts ;-)
Could you please look at this.
I need to add an additional load forecasting method that I'm working right now, is this guide is up-to-date?: https://github.com/davidusb-geek/emhass/blob/master/docs/develop.md#adding-a-parameter

GeoDerp, 2024-12-25:
> @GeoDerp I tried to look at this but got a bit lost in those JS cripts ;-)
> Could you please look at this?
> I need to add an additional load forecasting method that I'm working right now, is this guide is up-to-date?: https://github.com/davidusb-geek/emhass/blob/master/docs/develop.md#adding-a-parameter
> 

I believe this was fixed in 11.3 *(one of the small things I fixed)*

Give me two secs and I'll check the docs.

davidusb-geek, 2024-12-25:
Thanks.
I actually transferred the issue because I tested on my fresh update of v0.11.4 and the mlforecaster option was not there. Maybe I needed to refresh the browser

GeoDerp, 2024-12-25:
> Thanks. I actually transferred the issue because I tested on my fresh update of v0.11.4 and the mlforecaster option was not there. Maybe I needed to refresh the browser

Just checked on my EMHASS v0.11.4, seems like its there, so im not sure. Maybe it is some sort of caching thing?

overas, 2024-12-26:
Home Assistent EMHASS v0.11.4 addon. I can now choose 'mlforecast' in Load forecast method with the web GUI and I can confirm it works. For me you can close this issue.

[image]

closed 2024-12-30 by davidusb-geek