davidusb-geek/emhass #404: Feature request: MPC Solcast use more than 1 day data

BDVGitHub, 2024-12-26:
When at MPC controller the forecast horizon is longer than 1 day. Is it possible to then take into account the Solcast forecast of more than 1 day, since with Solcast you get a 30-minute forecast for up to the next 3 days? Or is this already the case?
[image]
File with Solcast data in json format [Solcast_Forecasts.json](https://github.com/user-attachments/files/18254305/Solcast_Forecasts.json)

GeoDerp, 2025-01-27:
Greetings @BDVGitHub ,
yep you should be able to. try setting `historic_days_to_retrieve ` to like `3`.
Alternatively if you [cache solcast](https://emhass.readthedocs.io/en/latest/forecasts.html#caching-solcast) you will automatically have `historic_days_to_retrieve*2` of solcast data cached in EMHASS

BDVGitHub, 2025-01-27:
Hey @GeoDerp 
In this case, it is not the dates of the days that have passed (historical dates) but the forecast of the days ahead that interests me.
Becuase you are talking about `**historic**_days_to_retrieve` ...

GeoDerp, 2025-01-27:
I totally get your confusion and thats more my bad for the english of that parameter.
Try passing `historic_days_to_retrieve: 2` as a runtime parameter and see what it comes back with

GeoDerp, 2025-01-27:
Actually hang on, I reject my statement. Let me look at the code some more.

GeoDerp, 2025-01-27:
I think your right, and there is more work to be done here. 
@davidusb-geek 
Im thinking of creating a `forecast_days_to_retrieve` param. so we have one for days of past data to retrieve and days of forecast / predicted data.

May look into this, this week, if I have some energy / time after work

purcell-lab, 2025-02-13:
I agree with this FR it would be good if the EMHASS native solcast would cache multiday future forecasts.

I am able to achieve this outcome by passing data from the solcast sensors as the runtime payload.

```yaml
        "pv_power_forecast": {{
          ([states('sensor.power_photovoltaics')|int(0)] +
          state_attr('sensor.solcast_pv_forecast_forecast_today', 'detailedForecast')|selectattr('period_start','gt',utcnow()) | map(attribute='pv_estimate')|map('multiply',1000)|map('int')|list +
          state_attr('sensor.solcast_pv_forecast_forecast_tomorrow', 'detailedForecast')|selectattr('period_start','gt',utcnow()) | map(attribute='pv_estimate')|map('multiply',1000)|map('int')|list +
          state_attr('sensor.solcast_pv_forecast_forecast_day_3', 'detailedForecast')|selectattr('period_start','gt',utcnow()) | map(attribute='pv_estimate')|map('multiply',1000)|map('int')|list +          
          state_attr('sensor.solcast_pv_forecast_forecast_day_4', 'detailedForecast')|selectattr('period_start','gt',utcnow()) | map(attribute='pv_estimate')|map('multiply',1000)|map('int')|list
          )| tojson
        }},
```

[image]

MMicieli, 2026-10-10:
@davidusb-geek, this issue can be closed as resolved by #937, merged on 6 June 2026.

closed 2026-10-10 by davidusb-geek