davidusb-geek/emhass #413: New method to deal with HA data results in IndexError: list index out of range

meks007, 2024-12-31:
https://github.com/davidusb-geek/emhass/blob/c946b385ed36b6cc3fdc1b16ec1d099142012560/src/emhass/utils.py#L184 
introduced with commit https://github.com/davidusb-geek/emhass/commit/54638b9b94b52fc42017a5bd17ab720440a9010b 

results in the following error.

```
2024-12-31 00:00:04,222 - web_server - ERROR - Exception on /action/dayahead-optim [POST]
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 1511, in wsgi_app
    response = self.full_dispatch_request()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 919, in full_dispatch_request
    rv = self.handle_user_exception(e)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 917, in full_dispatch_request
    rv = self.dispatch_request()
         ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 902, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/emhass/web_server.py", line 388, in action_call
    input_data_dict = set_input_data_dict(
                      ^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/emhass/command_line.py", line 107, in set_input_data_dict
    params = utils.update_params_with_ha_config(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/emhass/utils.py", line 184, in update_params_with_ha_config
    params['passed_data']['custom_predicted_temperature_id'][k].update(
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^
IndexError: list index out of range
```

davidusb-geek, 2025-01-04:
What are the data inputs that you are using to produce this error?

meks007, 2025-01-04:
Complete log output is below. Worth mentioning is that I've set the number of deferrable loads to 0 in config (which you can't in UI), because I want to submit 0-n loads via runtime params. In https://github.com/davidusb-geek/emhass/blob/c946b385ed36b6cc3fdc1b16ec1d099142012560/src/emhass/utils.py#L184 you don't check if k is >=1 since UI doesn't allow 0 loads.

> 2025-01-04 00:00:03,206 - web_server - INFO - Passed runtime parameters: {'nominal_power_of_deferrable_loads': [1200, 2000, 1500, 4000, 1400, 2000, 10800], 'operating_hours_of_each_deferrable_load': [0, 0, 0, 1, 4, 1, 0], 'treat_deferrable_load_as_semi_cont': [True, True, True, True, True, True, False], 'set_deferrable_load_single_constant': [True, True, True, True, True, True, False], 'start_timesteps_of_each_deferrable_load': [0, 5, 7, 10, 0, 0, 0], 'end_timesteps_of_each_deferrable_load': [0, 17, 18, 16, 0, 0, 0], 'custom_deferrable_forecast_id': [{'entity_id': 'sensor.p_deferrable_dishwasher', 'unit_of_measurement': 'W', 'friendly_name': 'Geschirrspüler (Optimiert)'}, {'entity_id': 'sensor.p_deferrable_washingmachine', 'unit_of_measurement': 'W', 'friendly_name': 'Waschmaschine (Optimiert)'}, {'entity_id': 'sensor.p_deferrable_dryer', 'unit_of_measurement': 'W', 'friendly_name': 'Wäschetrockner (Optimiert)'}, {'entity_id': 'sensor.p_deferrable_cooking', 'unit_of_measurement': 'W', 'friendly_name': 'Kochen (Optimiert)'}, {'entity_id': 'sensor.p_deferrable_heating_dhwhc_overshoot', 'unit_of_measurement': 'W', 'friendly_name': 'Überheizen (Optimiert)'}, {'entity_id': 'sensor.p_deferrable_heating_dhw', 'unit_of_measurement': 'W', 'friendly_name': 'Warmwasser (Optimiert)'}, {'entity_id': 'sensor.p_deferrable_car', 'unit_of_measurement': 'W', 'friendly_name': 'Auto (Optimiert)'}], 'entity_save': True, 'publish_prefix': 'dh_', 'number_of_deferrable_loads': 7}
2025-01-04 00:00:03,206 - web_server - INFO -  >> Setting input data dict
2025-01-04 00:00:03,206 - web_server - INFO - Setting up needed data
2025-01-04 00:00:03,208 - web_server - INFO - Retrieving weather forecast data using method = csv
2025-01-04 00:00:03,211 - web_server - INFO - Retrieving data from hass for load forecast using method = naive
2025-01-04 00:00:03,212 - web_server - INFO - Retrieve hass get data method initiated...
2025-01-04 00:00:12,645 - web_server - INFO -  >> Performing dayahead optimization...
2025-01-04 00:00:12,645 - web_server - INFO - Performing day-ahead forecast optimization
2025-01-04 00:00:12,654 - web_server - INFO - Perform optimization for the day-ahead
2025-01-04 00:00:12,798 - web_server - INFO - Status: Optimal
2025-01-04 00:00:12,798 - web_server - INFO - Total value of the Cost function = -2.28
2025-01-04 00:00:12,804 - web_server - INFO - Publishing data to HASS instance
2025-01-04 00:00:12,809 - web_server - INFO - Successfully posted to sensor.dh_p_pv_forecast = 0.0
2025-01-04 00:00:12,812 - web_server - INFO - Successfully posted to sensor.dh_p_load_forecast = 282.38
2025-01-04 00:00:12,814 - web_server - INFO - Successfully posted to sensor.dh_p_deferrable_dishwasher = 0.0
2025-01-04 00:00:12,817 - web_server - INFO - Successfully posted to sensor.dh_p_deferrable_washingmachine = 0.0
2025-01-04 00:00:12,819 - web_server - INFO - Successfully posted to sensor.dh_p_deferrable_dryer = 0.0
2025-01-04 00:00:12,822 - web_server - INFO - Successfully posted to sensor.dh_p_deferrable_cooking = 0.0
2025-01-04 00:00:12,825 - web_server - INFO - Successfully posted to sensor.dh_p_deferrable_heating_dhwhc_overshoot = 0.0
2025-01-04 00:00:12,827 - web_server - INFO - Successfully posted to sensor.dh_p_deferrable_heating_dhw = 0.0
2025-01-04 00:00:12,829 - web_server - INFO - Successfully posted to sensor.dh_p_deferrable_car = 0.0
2025-01-04 00:00:12,831 - web_server - INFO - Successfully posted to sensor.dh_p_grid_forecast = 282.38
2025-01-04 00:00:12,833 - web_server - INFO - Successfully posted to sensor.dh_total_cost_fun_value = -2.28
2025-01-04 00:00:12,834 - web_server - INFO - Successfully posted to sensor.dh_optim_status = Optimal
2025-01-04 00:00:12,836 - web_server - INFO - Successfully posted to sensor.dh_unit_load_cost = 0.142
2025-01-04 00:00:12,837 - web_server - INFO - Successfully posted to sensor.dh_unit_prod_price = 0.0973
2025-01-04 00:00:12,985 - web_server - INFO - Passed runtime parameters: {'publish_prefix': 'dh_'}
2025-01-04 00:00:12,985 - web_server - INFO -  >> Setting input data dict
2025-01-04 00:00:12,985 - web_server - INFO - Setting up needed data
2025-01-04 00:00:12,986 - web_server - INFO -  >> Publishing data...
2025-01-04 00:00:12,986 - web_server - INFO - Publishing data to HASS instance
2025-01-04 00:00:12,996 - web_server - INFO - Successfully posted to sensor.dh_p_deferrable_car = 0
2025-01-04 00:00:13,008 - web_server - INFO - Successfully posted to sensor.dh_p_deferrable_dryer = 0
2025-01-04 00:00:13,016 - web_server - INFO - Successfully posted to sensor.dh_optim_status = Optimal
2025-01-04 00:00:13,027 - web_server - INFO - Successfully posted to sensor.dh_unit_prod_price = 0.0973
2025-01-04 00:00:13,036 - web_server - INFO - Successfully posted to sensor.dh_p_pv_forecast = 0
2025-01-04 00:00:13,046 - web_server - INFO - Successfully posted to sensor.dh_p_deferrable_heating_dhwhc_overshoot = 0
2025-01-04 00:00:13,059 - web_server - INFO - Successfully posted to sensor.dh_p_grid_forecast = 282.38
2025-01-04 00:00:13,070 - web_server - INFO - Successfully posted to sensor.dh_unit_load_cost = 0.142
[... cut ...]
2025-01-04 00:00:13,092 - web_server - INFO - Successfully posted to sensor.dh_p_deferrable_heating_dhw = 0
2025-01-04 00:00:13,105 - web_server - INFO - Successfully posted to sensor.dh_p_deferrable_washingmachine = 0
2025-01-04 00:00:13,113 - web_server - INFO - Successfully posted to sensor.dh_total_cost_fun_value = -2.28
2025-01-04 00:00:13,123 - web_server - INFO - Successfully posted to sensor.dh_p_load_forecast = 282.38
2025-01-04 00:00:13,135 - web_server - INFO - Successfully posted to sensor.dh_p_deferrable_cooking = 0

davidusb-geek, 2025-01-04:
Ok, get it. I will then make the code treat that special case

davidusb-geek, 2025-01-04:
We need to put this somewhere in the docs but the minimum number of deferrable loads is 1. But if you don't want any then set the number to 1 and put 0 hours to run that single deferrable load.
But I will still add that check

davidusb-geek, 2025-01-04:
Ok solved and merged in #418. 
But by default follow the procedure on my previous post.
Closing

closed 2025-01-04 by davidusb-geek

meks007, 2025-01-27:
Hey,

I tried adapting config.json to 1 load with 0 hours:

```
    "number_of_deferrable_loads": 1,
    "nominal_power_of_deferrable_loads": [
        11000
    ],
    "operating_hours_of_each_deferrable_load": [
        0
    ],
```

But the error is still happening in 0.12.4:

`[2025-01-27 23:44:34 +0100] [23] [INFO] Passed runtime parameters: {'nominal_power_of_deferrable_loads': [1200, 2000, 1500, 4000, 1400, 2000, 10800], 'operating_hours_of_each_deferrable_load': [1, 0, 0, 1, 4, 0, 0], 'treat_deferrable_load_as_semi_cont': [True, True, True, True, True, True, False], 'set_deferrable_load_single_constant': [True, True, True, True, True, True, False], 'start_timesteps_of_each_deferrable_load': [0, 6, 8, 11, 0, 0, 0], 'end_timesteps_of_each_deferrable_load': [0, 18, 19, 17, 0, 0, 0], 'custom_deferrable_forecast_id': [{'entity_id': 'sensor.p_deferrable_dishwasher', 'unit_of_measurement': 'W', 'friendly_name': 'Geschirrspüler (Optimiert)'}, {'entity_id': 'sensor.p_deferrable_washingmachine', 'unit_of_measurement': 'W', 'friendly_name': 'Waschmaschine (Optimiert)'}, {'entity_id': 'sensor.p_deferrable_dryer', 'unit_of_measurement': 'W', 'friendly_name': 'Wäschetrockner (Optimiert)'}, {'entity_id': 'sensor.p_deferrable_cooking', 'unit_of_measurement': 'W', 'friendly_name': 'Kochen (Optimiert)'}, {'entity_id': 'sensor.p_deferrable_heating_dhwhc_overshoot', 'unit_of_measurement': 'W', 'friendly_name': 'Überheizen (Optimiert)'}, {'entity_id': 'sensor.p_deferrable_heating_dhw', 'unit_of_measurement': 'W', 'friendly_name': 'Warmwasser (Optimiert)'}, {'entity_id': 'sensor.p_deferrable_car', 'unit_of_measurement': 'W', 'friendly_name': 'Auto (Optimiert)'}], 'entity_save': True, 'publish_prefix': 'dh_', 'number_of_deferrable_loads': 7}`

```
[2025-01-27 23:44:34 +0100] [23] [INFO]  >> Setting input data dict
[2025-01-27 23:44:34 +0100] [23] [INFO] Setting up needed data
[2025-01-27 23:44:34 +0100] [23] [ERROR] Exception on /action/dayahead-optim [POST]
Traceback (most recent call last):
  File "/app/.venv/lib/python3.11/site-packages/flask/app.py", line 1511, in wsgi_app
    response = self.full_dispatch_request()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/.venv/lib/python3.11/site-packages/flask/app.py", line 919, in full_dispatch_request
    rv = self.handle_user_exception(e)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/.venv/lib/python3.11/site-packages/flask/app.py", line 917, in full_dispatch_request
    rv = self.dispatch_request()
         ^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/.venv/lib/python3.11/site-packages/flask/app.py", line 902, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/src/emhass/web_server.py", line 414, in action_call
    input_data_dict = set_input_data_dict(
                      ^^^^^^^^^^^^^^^^^^^^
  File "/app/src/emhass/command_line.py", line 106, in set_input_data_dict
    params = utils.update_params_with_ha_config(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/src/emhass/utils.py", line 204, in update_params_with_ha_config
    params["passed_data"]["custom_predicted_temperature_id"][k].update(
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^
IndexError: list index out of range
```