davidusb-geek/emhass #422: Feature request: show where battery is charging from

molekuul, 2025-01-06:
In the optimization results, you see when the battery is charging, however you don’t see where the charging energy is provided by. Please add a output showing if the battery charging get it’s energy from the PV or from the GRID. This makes it easier to automate battery charging when you have a hybrid inverter, such as the growatt inverters.

nejcsu, 2025-01-08:
How do you instruct the inverter to go into charge/discharge/idle? I use a Deye inverter, and it looks like they only support six different time periods, where you can set charge/discharge parameters and limits.

radim2tej, 2025-01-09:
I'm using a Goodw GW10K-ET and an experimental integration that can charge and discharge in economy inverter mode. I solve the idle state by setting the depth of discharge to 95%. Since I don't know if the changes are written to the internal eeprom of the inverter, a check is made before each change to see if the change is necessary.
```
alias: EMHASS battery control
description: use battery by EMHASS
triggers:
  - trigger: state
    entity_id:
      - sensor.emhass_battery_control
  - trigger: state
    entity_id:
      - sensor.battery_state_of_charge
  - trigger: state
    entity_id:
      - sensor.soc_batt_forecast
  - trigger: state
    entity_id:
      - sensor.final_sell_kwh
conditions: []
actions:
  - alias: charge or idle battery_control for depth of discharge
    if:
      - condition: or
        conditions:
          - condition: state
            entity_id: sensor.emhass_battery_control
            state: charge
          - condition: state
            entity_id: sensor.emhass_battery_control
            state: idle
    then:
      - alias: disabling battery usage in charge and idle battery_control
        if:
          - alias: battery depth of discharge is below 95%
            condition: numeric_state
            entity_id: number.goodwe_maximum_vybiti_v_siti
            above: 5
        then:
          - alias: disable battery usage by setting the depth of discharge to 95%
            action: number.set_value
            metadata: {}
            data:
              value: 5
            target:
              entity_id: number.goodwe_maximum_vybiti_v_siti
    else:
      - alias: enable battery usage in normal and discharge battery_control
        if:
          - alias: battery depth of discharge is above 20%
            condition: numeric_state
            entity_id: number.goodwe_maximum_vybiti_v_siti
            below: 80
        then:
          - alias: enable battery use by setting the discharge depth to 20%
            action: number.set_value
            metadata: {}
            data:
              value: 80
            target:
              entity_id: number.goodwe_maximum_vybiti_v_siti
  - alias: overflow control at negative prices
    if:
      - condition: numeric_state
        entity_id: number.final_sell_kwh
        below: 0.2
    then:
      - alias: enable grid supply control
        if:
          - alias: supply control is off
            condition: state
            entity_id: switch.goodwe_rizeni_dodavky_do_site
[... 104 lines cut ...]
              - alias: inverter in eco_discharge mode
                action: select.select_option
                metadata: {}
                data:
                  option: eco_discharge
                target:
                  entity_id: select.goodwe_provozni_rezim_stridace
        else:
          - alias: disable battery charging/discharging by general mode
            if:
              - alias: inverter is in general mode
                condition: state
                entity_id: select.goodwe_provozni_rezim_stridace
                state: general
            then: []
            else:
              - alias: inverter in general mode
                action: select.select_option
                metadata: {}
                data:
                  option: general
                target:
                  entity_id: select.goodwe_provozni_rezim_stridace
mode: single
```

closed 2026-11-03 by davidusb-geek