davidusb-geek/emhass #425: Documentation: Docker volume for data storage

BDVGitHub, 2025-01-07:
Is it possible to add info about docker volume? So when Emhass is updated data is not lost. For more info see https://medium.com/@praveenadoni4456/understanding-docker-volumes-a-comprehensive-guide-with-examples-part-3-684851696695

GeoDerp, 2025-01-27:
Greeting @BDVGitHub ,
check out: https://emhass.readthedocs.io/en/latest/intro.html#docker-things-to-note

Somthing like this should help:
```
-v ~/emhass/config.json:/app/config.json
```
Which volume mounts your host file `~/emhass/config.json` to the docker container `/app/config.json`

BDVGitHub, 2025-01-27:
Would it possibly be possible to modify the Emhass Dockerfile to use a Docker volume by default?
In https://docs.docker.com/reference/dockerfile/ I found https://docs.docker.com/reference/dockerfile/#volume
So that when someone who does not know much about docker does not lose his data when updating Emhass, for example?

davidusb-geek, 2025-01-31:
That is not how it works. Its when you use the Dockerfile through a `docker run` command that you can mount volumes.
This is all done automatically if using the add-on, no data is lost between upgrades.
However if using the docker standalone method then you should set the volumes yourself and that is already well documented in the link provided previously: https://emhass.readthedocs.io/en/latest/intro.html#docker-things-to-note
Closing

closed 2025-01-31 by davidusb-geek