davidusb-geek/emhass #428: Possible problem with weight_battery_charge

radim2tej, 2025-01-18:
**Describe the problem**
I tried to make the battery amortization work using the weight_battery_charge parameter, setting it to 0.08 euro/kwh and leaving weight_battery_discharge at 0. The results are wrong. Then I investigated the LP constraint which is defined as: 
```
            objective = objective + plp.lpSum(
                -0.001
                * self.timeStep
                * (
                    self.optim_conf["weight_battery_discharge"] * P_sto_pos[i]
                    + self.optim_conf["weight_battery_charge"] * P_sto_neg[i]
                )
                for i in set_I
            )
```
To me, the value of weight_battery_charge must be negative, or the objective formulation must be (P_sto_neg is negative):
```
objective = objective + plp.lpSum(
                -0.001
                * self.timeStep
                * (
                    self.optim_conf["weight_battery_discharge"] * P_sto_pos[i]
                    - self.optim_conf["weight_battery_charge"] * P_sto_neg[i]
                )
                for i in set_I
            )
```
Please give me some thought if this is the case. I am using a hybrid inverter and battery.

radim2tej, 2025-01-21:
fix #430

davidusb-geek, 2025-01-21:
We oversaw this for a long time. Thanks for the fix.
I will take the time on the week-end to merge and release a new version

closed 2025-01-31 by davidusb-geek