davidusb-geek/emhass #431: Feature request: Exclusive deferrable loads

llevering, 2025-01-22:
It would be really neat if we could define that some deferrable load shouldn't be run at the same time.

Example: I have two heat pump based water heaters, that share the same air intake. When it is cold and both run at the same time it leads to defrosts cycles that can be avoided when they don't run at the same time.

Nice to have: maybe define it in such a way that deferrable either a completely exclusive (i.e. when two loads would cause a overcurrent in a certain circuit), or are just less efficient if run at the same time (leaving the model with more possible solutions).

P.s. Thanks for this great tool, really appreciate it!

davidusb-geek, 2025-01-22:
Yes this can be a nice little feature. We can add it to future development wish list.
For now there is maybe a possibility to easily achieve what you want (I have not tested it).
You could set an "artificial" very high value for the nominal power of the deferrable that you want to keep exclusive. This way the algorithm will try to allocate it in an exclusive manner. Just be careful to set value sufficiently high to achieve this, but not to high to avoid unfeasible conditions if for example the nominal power of the deferrable plus the current household power consumption is higher than the PV production plus the maximum power that can be imported from the grid.

llevering, 2025-01-24:
Yes that does work, though it ruïns the graph and cost estimate of course. But surely a very functional work-around!

meks007, 2025-02-10:
I've built a logic that does this before posting the deferrables as runtime params. To add another aspect of "exclusivity": dependency. I.e. you can't run a dryer before the washing machine is finished. So its not just exclusivity as in "not x at the same time as y" but also "y after x".

My suggestion is to add (1) an exclusive switch to a deferrable load, as well as (2) a priority rank and (3) a grouping logic. So you could say:

Washing machine: exclusive = false, priority = 10, group = washing
Dryer: exclusive = false, priority = 20, group = washing
Water heater 1: exclusive = false, priority = 10, group = heating
Water heater 2: exclusive = false, priority = 20, group = heating
Heating: exclusive = true, priority = 30, group = heating

* The logic is that all loads, regardless of their respective group, are checked for exclusivity. Exclusive loads are optimized in a way that they can run without any other load running at the same time.
* After that, all remaining loads within the same group are ranked based on priority and then optimized based on that rank.

In my example that logic would give the following ranked loads:

1. Heating
2. Water heater + Washing machine
3. Water heater 2 + Dryer

After ranking the usual runtime and load distribution logics are applied.