davidusb-geek/emhass #432: Suggestion: Add Docker for GitHub Copilot

BDVGitHub, 2025-01-27:
I see you are already using Dependabot to keep the dependencies up to date.
There is also a Copilot for Docker in GitHub, it could potentially be useful as the project uses Docker.
For example it can be use to help with finding project vulnerabilities through Docker Scout:.
The Docker extension also integrates with Docker Scout to surface a high-level summary of detected vulnerabilities and provide
the next steps to continue using Scout in your terminal via CLI: “@docker can you help me find vulnerabilities in my project?”
You can find the GitHub Docker Copilot here https://github.com/marketplace/docker-for-github-copilot

Related to https://github.com/davidusb-geek/emhass/issues/425 it might be useful to use docker volumes, so that no data is lost after updating Emhass.
See https://docs.docker.com/engine/storage/volumes/ for more information.

GeoDerp, 2025-01-27:
Greetings @BDVGitHub, thanks for your suggestion,
we do currently have  scanning from the following Github Actions:

Source composition analysis (checks docker dependencies for known vulnerabilities): [Google OSV](https://github.com/davidusb-geek/emhass/actions/workflows/publish_docker-test.yaml)
Static app security testing (checks source code): [CodeQL](https://github.com/davidusb-geek/emhass/actions/workflows/codeql.yml)

However we don't currently have a action to check PRs for as a code review / code recommendation (something im wondering if copilot could help with ).

Is there something im missing (if so let me know), and also is copilots implementation and action open source?

BDVGitHub, 2025-01-27:
As far as I understand it, copilot can be used for code recommendation.

davidusb-geek, 2025-01-29:
I think that this is out of scope. I don't see how we could possibly add copilot on a workflow, it does not makes sense to me.

closed 2025-01-29 by davidusb-geek