davidusb-geek/emhass #436: Feature Request: Use Docker Hub Builder

BDVGitHub, 2025-01-27:
Is it possible to use [Docker Hub Builder](https://docs.docker.com/docker-hub/repos/manage/builds/link-source/) so that when an new Emhass relaese is published its also automatic available on Docker Hub?
On [Docker Hub the latest relaese is 0.10.6](https://hub.docker.com/r/davidusb/emhass-docker-standalone/tags) while the latest GitHub relaese is 0.12.3

[image]

davidusb-geek, 2025-01-27:
No, we dropped docker hub builds some months ago to change to the official HA image repository: ghcr.io/home-assistant

closed 2025-01-27 by davidusb-geek

BDVGitHub, 2025-01-28:
The link is not working I get a 404 page error fault, is it possible to share the correct link? Because I don't know what you maen with official HA image repository