davidusb-geek/emhass #439: Reverting back support for skforecast v0.14.0

davidusb-geek, 2025-01-28:
> @GeoDerp This commit I think that it effectively broke the skforecast usage. It reverted it back to an old implementation.> 
> I think that you made this for it to be compatible with the uv package manager.> 
> We need to stick to the latest v0.14.0 for skforecast 

 _Originally posted by @davidusb-geek in [3ffde8a](https://github.com/davidusb-geek/emhass/commit/3ffde8a9ac67f5c6884b5042810805d5eb4d114a#commitcomment-151873712)_

davidusb-geek, 2025-01-28:
@GeoDerp Could you please help on this?

GeoDerp, 2025-01-28:
Yep, Will look into this tonight.

GeoDerp, 2025-01-28:
The reverse bump was my bad, This was when I was trying to get arm working.

davidusb-geek, 2025-01-28:
Yes no problem. The correct code is there in the history anyways

closed 2025-02-01 by davidusb-geek