davidusb-geek/emhass #444: Follow-Up #413 : New method to deal with HA data results in IndexError: list index out of range

meks007, 2025-02-02:
Hey,

following up on closed #413. The error still appears in 0.12.4. As I've mentioned in the last comment I've set the config to have 1 deferrable load with 0 hours. EMHASS still fails dayahead-optim with the following error.

Is there anything I'm doing wrong?

```
[2025-01-27 23:44:34 +0100] [23] [INFO]  >> Setting input data dict
[2025-01-27 23:44:34 +0100] [23] [INFO] Setting up needed data
[2025-01-27 23:44:34 +0100] [23] [ERROR] Exception on /action/dayahead-optim [POST]
Traceback (most recent call last):
  File "/app/.venv/lib/python3.11/site-packages/flask/app.py", line 1511, in wsgi_app
    response = self.full_dispatch_request()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/.venv/lib/python3.11/site-packages/flask/app.py", line 919, in full_dispatch_request
    rv = self.handle_user_exception(e)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/.venv/lib/python3.11/site-packages/flask/app.py", line 917, in full_dispatch_request
    rv = self.dispatch_request()
         ^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/.venv/lib/python3.11/site-packages/flask/app.py", line 902, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/src/emhass/web_server.py", line 414, in action_call
    input_data_dict = set_input_data_dict(
                      ^^^^^^^^^^^^^^^^^^^^
  File "/app/src/emhass/command_line.py", line 106, in set_input_data_dict
    params = utils.update_params_with_ha_config(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/src/emhass/utils.py", line 204, in update_params_with_ha_config
    params["passed_data"]["custom_predicted_temperature_id"][k].update(
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^
IndexError: list index out of range
```

The payload I pass is:

```
{
   "nominal_power_of_deferrable_loads":[
      1200,
      2000,
      1500,
      4000,
      1400,
      2000,
      10800
   ],
   "operating_hours_of_each_deferrable_load":[
      0,
      0,
      0,
      1,
      4,
      1,
      0
   ],
   "treat_deferrable_load_as_semi_cont":[
      true,
      true,
      true,
      true,
      true,
      true,
      false
   ],
   "set_deferrable_load_single_constant":[
      true,
      true,
      true,
      true,
      true,
      true,
      false
   ],
   "start_timesteps_of_each_deferrable_load":[
      0,
      10,
      12,
      15,
      0,
      0,
      0
   ],
   "end_timesteps_of_each_deferrable_load":[
      0,
      22,
      23,
      21,
      0,
      0,
      0
   ],
   "custom_deferrable_forecast_id":[
      {
         "entity_id":"sensor.p_deferrable_dishwasher",
         "unit_of_measurement":"W",
         "friendly_name":"Geschirrspüler (Optimiert)"
      },
      {
         "entity_id":"sensor.p_deferrable_washingmachine",
         "unit_of_measurement":"W",
         "friendly_name":"Waschmaschine (Optimiert)"
      },
      {
         "entity_id":"sensor.p_deferrable_dryer",
         "unit_of_measurement":"W",
         "friendly_name":"Wäschetrockner (Optimiert)"
      },
      {
         "entity_id":"sensor.p_deferrable_cooking",
         "unit_of_measurement":"W",
         "friendly_name":"Kochen (Optimiert)"
      },
      {
         "entity_id":"sensor.p_deferrable_heating_dhwhc_overshoot",
         "unit_of_measurement":"W",
         "friendly_name":"Überheizen (Optimiert)"
      },
      {
         "entity_id":"sensor.p_deferrable_heating_dhw",
         "unit_of_measurement":"W",
         "friendly_name":"Warmwasser (Optimiert)"
      },
      {
         "entity_id":"sensor.p_deferrable_car",
         "unit_of_measurement":"W",
         "friendly_name":"Auto (Optimiert)"
      }
   ],
   "entity_save":true,
   "publish_prefix":"dh_",
   "number_of_deferrable_loads":7
}
```

I've then tried to add custom_predicted_temperature_id to the payload:

```
   "custom_predicted_temperature_id":[
      null,
      null,
      null,
      null,
      null,
      null,
      null
   ]
```

That leads to this error though:

```
[2025-02-02 19:53:54 +0100] [23] [ERROR] Exception on /action/dayahead-optim [POST]
Traceback (most recent call last):
  File "/app/.venv/lib/python3.11/site-packages/flask/app.py", line 1511, in wsgi_app
    response = self.full_dispatch_request()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/.venv/lib/python3.11/site-packages/flask/app.py", line 919, in full_dispatch_request
    rv = self.handle_user_exception(e)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/.venv/lib/python3.11/site-packages/flask/app.py", line 917, in full_dispatch_request
    rv = self.dispatch_request()
         ^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/.venv/lib/python3.11/site-packages/flask/app.py", line 902, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/src/emhass/web_server.py", line 414, in action_call
    input_data_dict = set_input_data_dict(
                      ^^^^^^^^^^^^^^^^^^^^
  File "/app/src/emhass/command_line.py", line 106, in set_input_data_dict
    params = utils.update_params_with_ha_config(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/src/emhass/utils.py", line 204, in update_params_with_ha_config
    params["passed_data"]["custom_predicted_temperature_id"][k].update(
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'update'
```

davidusb-geek, 2025-02-02:
Ok I'm a bit lost.
You say in the closed issue that you're setting this in the config.json:

```
    "number_of_deferrable_loads": 1,
    "nominal_power_of_deferrable_loads": [
        11000
    ],
    "operating_hours_of_each_deferrable_load": [
        0
    ],
```

So how come that you're then trying to pass in your payload lists with 7 values?
Just don't pass this payload. By doing so the system is inferring that you have 7 deferrable loads

meks007, 2025-02-02:
that being said i've worked around it by adding the following object to my payload. that way it appears to be working.
please do advise in case i misunderstood and should do anything differently.

```
"custom_predicted_temperature_id":[
   {
      "emhass_issue":"#444"
   },
   {
      "emhass_issue":"#444"
   },
   {
      "emhass_issue":"#444"
   },
   {
      "emhass_issue":"#444"
   },
   {
      "emhass_issue":"#444"
   },
   {
      "emhass_issue":"#444"
   },
   {
      "emhass_issue":"#444"
   }
```

davidusb-geek, 2025-02-02:
> because i'm determining loads programatically and then submit a the mentioned json payload depending on which loads need optimization. that's also what the documentation suggests when dealing with dynamic loads.
> 
> example: water doesnt need heating every day, only every other day. car doesnt need charging every day, only when its below a certain charge, dishwasher only needs to run every 4 days, washing machine once a week et cetera.
> 
> so submitting dynamic loads is crucial to have EMHASS do the correct optimization. not submitting loads at runtime is not an option.

Ok I think I'm getting there trying to understand this.
Yes you can set load programatically but then don't set `"number_of_deferrable_loads": 1`.
What you can do is set `"number_of_deferrable_loads": 7` and then just set the `"operating_hours_of_each_deferrable_load"` programatically for each of your loads. If you want to deactivate some loads then set that to 0. 
But maybe you're already doing that?
It may be a bug in the need of setting the `"custom_predicted_temperature_id"` fields. If you're not setting any thermal load you should not need to do this.
I'll look into this...

meks007, 2025-02-02:
but setting the number is exactly what the documentation says: 
https://emhass.readthedocs.io/en/latest/intro.html#passing-other-data-at-runtime

check your code, youre not checking if a load is a thermal load, hence the error.

davidusb-geek, 2025-02-02:
> but setting the number is exactly what the documentation says:
> https://emhass.readthedocs.io/en/latest/intro.html#passing-other-data-at-runtime

It is possible to set it with payloads, but then the `number_of_deferrable_loads` should be equal to the length os the passed lists for the deferrable loads settings, ex `operating_hours_of_each_deferrable_load`.

> check your code, youre not checking if a load is a thermal load, hence the error.

Yes, will search for this bug

meks007, 2025-02-02:
> It is possible to set it with payloads, but then the `number_of_deferrable_loads` should be equal to the length os the passed lists for the deferrable loads settings, ex `operating_hours_of_each_deferrable_load`.

yep, if you check my initial comment my payload has exactly 7 entries for each object, 1 entry for each deferrable load. that aligns perfectly with number_of_deferrable_loads. so it should work as per your definition, but it doesn't because there's a check for thermal loads missing in utils.py:204

thx for looking into it

davidusb-geek, 2025-02-02:
Ok this was just some bad coding.
Fixed on #445

davidusb-geek, 2025-02-02:
Please update to v0.12.5 and check if this is solved. It should be.
If so please close this issue

meks007, 2025-02-03:
fixed, thank you!

closed 2025-02-03 by meks007