davidusb-geek/emhass #446: BUG v12.5 No module named 'skforecast.ForecasterAutoreg

mime24, 2025-02-03:
**Describe the bug**
since updated to 12.5 my MPC fails due to issue "No module named 'skforecast.ForecasterAutoreg"

**To Reproduce**
rund MPC with runtime parameters as shown below

**Expected behavior**
should work as 12.4 and just execute the MPC

**Screenshots**

```

[2025-02-03 13:00:01 +0100] [23] [INFO] Passed runtime parameters: {'load_forecast_method': 'mlforecaster', 'prediction_horizon': 11, 'entity_save': True, 'publish_prefix': 'emhass_', 'soc_init': 0.23800000000000002, 'soc_final': 0.7, 'load_cost_forecast': [0.3676, 0.3781, 0.4039, 0.4335, 0.4874, 0.4692, 0.4697, 0.43, 0.4124, 0.402, 0.392], 'operating_hours_of_each_deferrable_load': [0, 2, 0, 0], 'def_current_state': [False, True, False, True], 'end_timesteps_of_each_deferrable_load': [0, 1, 0, 0], 'nominal_power_of_deferrable_loads': [1600, 300, 1000, 1466.0], 'def_load_config': [{}, {}, {}, {'thermal_config': {'heating_rate': 1.42, 'cooling_constant': 0.021, 'overshoot_temperature': 22.5, 'start_temperature': 21.0, 'desired_temperatures': [21, 21, 21, 21, 21, 21, 21, 21, 21, 21, 21]}}], 'outdoor_temperature_forecast': [0.8, 1.5, 2.0, 2.0, 1.9, -0.1, -1.6, -1.7, -1.9, -2.1, -2.4]}
[2025-02-03 13:00:01 +0100] [23] [INFO]  >> Setting input data dict
[2025-02-03 13:00:01 +0100] [23] [INFO] Setting up needed data
[2025-02-03 13:00:01 +0100] [23] [INFO] Retrieve hass get data method initiated...
[2025-02-03 13:00:15 +0100] [23] [INFO] Retrieving weather forecast data using method = scrapper
[2025-02-03 13:00:20 +0100] [23] [INFO] Retrieving data from hass for load forecast using method = mlforecaster
[2025-02-03 13:00:20 +0100] [23] [INFO] Retrieve hass get data method initiated...
[2025-02-03 13:00:44 +0100] [23] [ERROR] Exception on /action/naive-mpc-optim [POST]
Traceback (most recent call last):
  File "/app/.venv/lib/python3.12/site-packages/flask/app.py", line 1511, in wsgi_app
    response = self.full_dispatch_request()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/.venv/lib/python3.12/site-packages/flask/app.py", line 919, in full_dispatch_request
    rv = self.handle_user_exception(e)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/.venv/lib/python3.12/site-packages/flask/app.py", line 917, in full_dispatch_request
    rv = self.dispatch_request()
         ^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/.venv/lib/python3.12/site-packages/flask/app.py", line 902, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/src/emhass/web_server.py", line 414, in action_call
    input_data_dict = set_input_data_dict(
                      ^^^^^^^^^^^^^^^^^^^^
  File "/app/src/emhass/command_line.py", line 282, in set_input_data_dict
    P_load_forecast = fcst.get_load_forecast(
                      ^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/src/emhass/forecast.py", line 1124, in get_load_forecast
    mlf = pickle.load(inp)
          ^^^^^^^^^^^^^^^^
ModuleNotFoundError: No module named 'skforecast.ForecasterAutoreg'
```

**Home Assistant installation type**
 - Home Assistant OS

**Your hardware**
- OS: HA OS
- Architecture: Homeassistant Yellow, RaspberryPi CM4, aarch64

**EMHASS installation type**
 - Add-on

davidusb-geek, 2025-02-03:
You just need to erase the saved *pkl files containing your previously saved ML forecasting models. Then relaunch the new ML models training. Then relaunch MPC.

mime24, 2025-02-03:
thank you.
it was actually enough to the train model again.

maybe you can incorporate a kind of info message or at least a note in the changelog, for other users.

closed 2025-02-03 by mime24