davidusb-geek/emhass #451: FileNotFoundError: [Errno 2] No such file or directory: ‘/app/data/data_load_cost_forecast.csv’

davidusb-geek, 2025-02-07:
In the community forum the user @Brkie reported this issue:

> Hello,
> will someone explain to me where his directory is? I am usign addon; there is no files in addon_config related to emhass, nothing in addon folder, nothing in config folder or share folder… My error message:
> FileNotFoundError: [Errno 2] No such file or directory: ‘/app/data/data_load_cost_forecast.csv’

**Describe the bug**
No description of when the bug arises: dayhead optim, MPC optim, publish task, ML train, etc.

**To Reproduce**
NA

**Screenshots**
```
FileNotFoundError: [Errno 2] No such file or directory: ‘/app/data/data_load_cost_forecast.csv’
```

**Home Assistant installation type**
 - Home Assistant OS (or so it seems)

**Your hardware**
- OS: HA OS
- Architecture: amd64 (not sure)

**EMHASS installation type**
 - Add-on

davidusb-geek, 2025-02-07:
@GeoDerp Could it be possible to take a look at this when you have some spare time?
It seems like a bug to me.
In the old days we were copying these type of files "manually" in the Dockerfile but now I'm a bit lost with our current version of the Dockerfile, it exceed my skills

Brkie, 2025-02-07:
> In the community forum the user [@Brkie](https://github.com/Brkie) reported this issue:
>
>
> **Describe the bug** No description of when the bug arises: dayhead optim, MPC optim, publish task, ML [...]

Here comes more info if it is of any help:

Version | core-2025.1.4
-- | --
Installationstyp | Home Assistant OS
Utveckling | false
Supervisor | true
Docker | true
Användare | root
Virtuell miljö | false
Python-version | 3.13.1
Operativsystemfamilj | Linux
Operativsystemversion | 6.6.73-haos
Processorarkitektur | x86_64
Tidszon | Europe/Stockholm
Konfigurationskatalog | /config

amd64 is correct

davidusb-geek, 2025-02-07:
Thanks for the info
The issue is solved as per PR #452
It will be effective on the next release

closed 2025-02-07 by davidusb-geek

GeoDerp, 2025-02-08:
100% works for me