davidusb-geek/emhass #454: Can't connect to Home Assistant due to user error

patbro, 2025-02-08:
**Describe the bug**
EMHASS can't connect to Home Assistant, like not at all, due to user error.

**To Reproduce**
Follow the steps to run EMHASS in a Docker container as explained in Method 2.

I used Docker compose in order to get EMHASS running as follows:
```
  emhass:
    image: ghcr.io/davidusb-geek/emhass:v0.12.5
    container_name: emhass
    environment:
    - TZ=Europe/Amsterdam
    ports:
    - 5000:5000
    volumes:
      - type: bind
        source: ./volumes/emhass/secrets_emhass.yml
        target: /app/secrets_emhass.yaml
        read_only: true
    restart: unless-stopped
```

`secrets_emhass.yml` contains the following:
```
hass_url: http://192.168.2.28:8123
long_lived_token: eyJhREDACTEDbys
time_zone: Europe/Amsterdam
```

**Expected behavior**
Browsing to `192.168.2.28:5000` and clicking 'Day-ahead optimization' should yield some result other than can't connect to Home Assistant.

**Home Assistant installation type**
 - Home Assistant Core via Docker compose

**Your hardware**
- Raspberry Pi 4 (4 GB) running: `Linux raspberrypi 6.6.31+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux`

**EMHASS installation type**
- Docker Standalone

**Resolution**
Adding a final slash to the `hass_url` like `http://192.168.2.28:8123/` after checking the [source code](https://github.com/davidusb-geek/emhass/blob/master/src/emhass/retrieve_hass.py#L97), made EMHASS successfully connect to Home Assistant. 

Suggestion: change line 97 to `/api/config` instead of `api/config`, and/or validate the URL instead prior to using it (whether it ends with a backslash?).

davidusb-geek, 2025-02-09:
Hi, yes we can add a check to test if yor passed url finish with a slash

davidusb-geek, 2025-02-09:
Proposed a fix to this here: https://github.com/davidusb-geek/emhass/commit/76b823c2541f000d9eee3563194c8aa85b208aed

closed 2025-02-09 by davidusb-geek