davidusb-geek/emhass #455: Security vulnerabilities with docker-build-test.yaml

davidusb-geek, 2025-02-09:
@GeoDerp 
I have seen an important number of vulnerabilities linked to this pipeline, specifically on the `osv-scan` method.
This uses a branch of yours in:
```
uses: "geoderp/osv-scanner-action/.github/workflows/osv-scanner-reusable.yml@v0.0.1"
```
Don't you think we should switch to the updated original `osv-scanner-action`?

GeoDerp, 2025-02-09:
> @GeoDerp
> I have seen an important number of vulnerabilities linked to this pipeline, specifically on the `osv-scan` method.
> This uses a branch of yours in:
> ```
> uses: [...]

100% , they recently merged my pull request 😁

davidusb-geek, 2025-02-16:
Preparing the upcoming release, I've just comment out the osv-scan part on `docker-build-test.yaml`, waiting until they release the new improved version.

GeoDerp, 2025-02-16:
> Preparing the upcoming release, I've just comment out the osv-scan part on `docker-build-test.yaml`, waiting until they release the new improved version.

Once they release I'll probably do a review and see what might need to be changed

GeoDerp, 2025-02-16:
PS. The errors in the actions are because the are known vulnerabilities detected in the Linux container. Unfortunately from what I have seen the known vulnerabilities are related to the Linux version of debian.

closed 2025-04-21 by davidusb-geek