davidusb-geek/emhass #459: hardcoded sensor_power_load_no_var_loads sensor name in utils.py

scruysberghs, 2025-02-12:
**Describe the bug**
A clear and concise description of what the bug is.
Whet you set a custom sensor for sensor_power_load_no_var_loads in the the EMHASS config file:
you still get the error :

`ERROR - emhass.web_server - The retrieved JSON is empty, A sensor:sensor.power_load_no_var_loads may have 0 days of history, passed sensor may not be correct, or days to retrieve is set too heigh`

**To Reproduce**
Steps to reproduce the behavior
Set a non default sensorname in the configfile and run a mpc or day ahead optimizer with passing the non default sensorname as runtime parameter

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Home Assistant installation type**
 - Home Assistant OS
 - Home Assistant Supervised
 - Home Assistant Core

**Your hardware**
- OS: HA OS, Windows, Linux, etc
- Architecture: amd64, armhf, armv7, aarch64

**EMHASS installation type**
 - Add-on, Docker Standalone, Legacy Python virtual environment

**Additional context**
Add any other context about the problem here.

Created a PR for what where I think the problem is, line 615 in utils.py reads 
`  if "var_model" not in runtimeparams.keys():
            var_model = "sensor.power_load_no_var_loads"`
while it should get the sensorname from the config file

davidusb-geek, 2025-02-12:
It makes much more sense to go fetch the sensor name on the configuration. 
I don't know why this was hard coded like that.
Thanks for the PR

davidusb-geek, 2025-02-17:
Merged and released on v0.12.7