davidusb-geek/emhass #464: Failed to install to install add-on on RPi 3B+

768g, 2025-02-17:
When attempting to install Emhass v0.12.6 or v0.12.7 on a Raspberry Pi 3B+ I get the following error.

Failed to install add-on
Can't install ghcr.io/davidusb-geek/emhass:v0.12.7: 404 Client Error for http+docker://localhost/v1.47/images/ghcr.io/davidusb-geek/emhass:v0.12.7/json: Not Found ("No such image: ghcr.io/davidusb-geek/emhass:v0.12.7")

Is the build process creating the docker files for RPi?

davidusb-geek, 2025-02-17:
Yes but only for aarch64 based boards

768g, 2025-02-18:
This last worked for me on v0.12.2 or v0.12.3 on 32bit RPi 3B+. Is 32 bit no longer supported?

davidusb-geek, 2025-02-18:
I'm afraid not.
But if a solution is found by someone we could reinstate builds on these older systems.
@GeoDerp tried many times to adapt to these archs but at some point it became a real struggle.
@GeoDerp Maybe we can support again this by dropping `skforecast` just for these builds? I think that it was the main blocking factor because of `numba`. That would mean that people cannot use MLForecaster but at least they could run optimization with other forecast methods

GeoDerp, 2025-02-18:
I believe you can run 64bit on a pi3b? 
Will do some more digging soon.

GeoDerp, 2025-02-18:
[image]

May need to install 64bit version of haos?

davidusb-geek, 2025-04-21:
Closing as the only solution (unfortunately) is to switch to 64 bit OS.
Re-open if needed

closed 2025-04-21 by davidusb-geek

davidusb-geek, 2025-04-25:
Great! And it did installed fine?

768g, 2025-04-26:
> Great! And it did installed fine?

Yes, it did. Thank you.