davidusb-geek/emhass #470: Feature request: add device_class to published sensors

paulhomes, 2025-02-23:
I noticed when trying to do Home Assistant charts (the built-in ones) comparing power sensors from emhass with other power sensors that they end up on different charts and so are harder to compare as a result. I believe this is because the emhass power sensors don't include the device_class attribute whereas the other power sensors I have in HA do. I tested this by building a custom emhass docker image adding in device_class and publishing the sensors with a different prefix. Those sensors do end up on the same chart as the other non-emhass power sensors.

Here is an image to demonstrate:

[image]

The top chart shows the PV power forecast from emhass and the bottom chart shows the other power sensors and the PV power forecast from my custom emhass build that includes device_class.

What are your thoughts on adding the device_class attribute to the emhass published sensors? I'd be happy to submit a pull request if you think it would be a useful addition.

davidusb-geek, 2025-02-23:
Hi Paul, yes that would be an improvement.
Happy to accept a PR for this.

purcell-lab, 2025-02-25:
@paulhomes Great work on the PR.

Can this approach to updating the sensors in the registry also be used to supply the unique_id field in the registry as well?
https://github.com/davidusb-geek/emhass-add-on/issues/91

paulhomes, 2025-02-25:
I had a similar experience to that mentioned in issue 91. Last week I tried to use the HA REST API to post unique_id on something else I was working on and didn't have any success. It seems like you need to running inside HA using the native API to be able create a sensor with unique_id, or publish via something like MQTT.

There is a suggestion to pre-define the sensors in yaml so the post is just updating existing sensors. I tried that and found I had to provide a state value in the yaml. That state value was updated when I posted but on HA restart went back to the yaml specified state. I ended up using an input_number to post to and a template sensor overlayed on top - it was messy and I wouldn't recommend it.

purcell-lab, 2025-02-25:
Thanks @paulhomes

davidusb-geek, 2025-02-27:
Let's close this as completed ;-)

closed 2025-02-27 by davidusb-geek

paulhomes, 2025-02-27:
Thanks @davidusb-geek for your help in getting this added.