davidusb-geek/emhass #478: Invalid escape sequence message in the log

g1za, 2025-02-27:
**Describe the bug**
After upgrading to 0.12.8 I get the following in the log:

[2025-02-27 23:56:27 +0100] [18] [INFO] Starting gunicorn 23.0.0
[2025-02-27 23:56:27 +0100] [18] [INFO] Listening at: http://0.0.0.0:5000 (18)
[2025-02-27 23:56:27 +0100] [18] [INFO] Using worker: gthread
[2025-02-27 23:56:27 +0100] [19] [INFO] Booting worker with pid: 19
/app/.venv/lib/python3.12/site-packages/pulp/tests/test_pulp.py:1776: SyntaxWarning: invalid escape sequence '\d'
  """
/app/.venv/lib/python3.12/site-packages/pulp/tests/test_pulp.py:1950: SyntaxWarning: invalid escape sequence '\d'
  command_line, option="strong", grp_pattern="\d+"
[2025-02-27 23:56:32 +0100] [19] [INFO] Obtaining parameters from config.json:

**To Reproduce**
upgrade to 0.12.8

**Home Assistant installation type**
 - Home Assistant OS

**Your hardware**
- OS: HA OS
- Architecture: virtualized under Proxmox

**EMHASS installation type**
 - Add-on

davidusb-geek, 2025-02-28:
Nothing to worry about and I don’t know what we can do about this one, as that message comes from inside the pulp package. So it's a small bug on their code on a part that is used to check that pulp install and solvers are correct

g1za, 2025-02-28:
Thanks David, closing it. ;)

closed 2025-02-28 by g1za