davidusb-geek/emhass #479: Runtime "delta_forecast_daily" not recognized

g1za, 2025-02-28:
**Describe the bug**
`delta_forecast_daily` runtime parameter is not recognized when running `dayahead-optim`. It still looks for the previous name of the param ( [https://emhass.readthedocs.io/en/latest/differences.html](url) )

If my command contains `delta_forecast_daily`I get the error below. If I rename it to `delta_forecast` no error is raised.

`[2025-02-28 21:01:31 +0100] [19] [INFO]  >> Obtaining params: 
[2025-02-28 21:01:31 +0100] [19] [INFO] Passed runtime parameters: {'pv_power_forecast': [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 140, 280, 428, 576, 699, 822, 911, 1000, 1055, 1110, 1114, 1118, 1080, 1042, 962, 881, 754, 627, 470, 314, 212, 110, 55, 0, 0, 0, 0], 'delta_forecast_daily': 1, 'number_of_deferrable_loads': 1, 'list_nominal_power_of_deferrable_loads': [0], 'list_operating_hours_of_each_deferrable_load': [1], 'list_treat_deferrable_load_as_semi_cont': [1], 'def_current_state': [0], 'list_set_deferrable_load_single_constant': [0], 'list_start_timesteps_of_each_deferrable_load': [0], 'list_end_timesteps_of_each_deferrable_load': [0], 'load_cost_forecast': [0.174425, 0.174425, 0.174425, 0.174425, 0.174425, 0.174425, 0.174425, 0.174425, 0.174425, 0.174425, 0.174425, 0.174425, 0.174425, 0.174425, 0.174425, 0.174425, 0.174425, 0.174425, 0.174425, 0.174425, 0.174425, 0.174425, 0.174425, 0.174425, 0.174425, 0.174425, 0.174425, 0.174425, 0.174425, 0.174425, 0.174425, 0.174425, 0.174425, 0.174425, 0.174425, 0.174425, 0.174425, 0.174425, 0.174425, 0.174425, 0.174425, 0.174425, 0.174425, 0.174425, 0.174425, 0.174425, 0.174425, 0.174425], 'prod_price_forecast': [0.146228, 0.146228, 0.14202, 0.14202, 0.130816, 0.130816, 0.135897, 0.135897, 0.131269, 0.131269, 0.127913, 0.127913, 0.127923, 0.127923, 0.12624, 0.12624, 0.128765, 0.128765, 0.141115, 0.141115, 0.145218, 0.145218, 0.14728, 0.14728, 0.147133, 0.147133, 0.134656, 0.134656, 0.131058, 0.131058, 0.118666, 0.118666, 0.110555, 0.110555, 0.118392, 0.118392, 0.132468, 0.132468, 0.143072, 0.143072, 0.153129, 0.153129, 0.15659, 0.15659, 0.156643, 0.156643, 0.151488, 0.151488]}
[2025-02-28 21:01:31 +0100] [19] [INFO]  >> Setting input data dict
[2025-02-28 21:01:31 +0100] [19] [INFO] Setting up needed data
[2025-02-28 21:01:31 +0100] [19] [ERROR] Exception on /action/dayahead-optim [POST]
Traceback (most recent call last):
  File "/app/.venv/lib/python3.12/site-packages/flask/app.py", line 1511, in wsgi_app
    response = self.full_dispatch_request()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/.venv/lib/python3.12/site-packages/flask/app.py", line 919, in full_dispatch_request
    rv = self.handle_user_exception(e)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/.venv/lib/python3.12/site-packages/flask/app.py", line 917, in full_dispatch_request
    rv = self.dispatch_request()
         ^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/.venv/lib/python3.12/site-packages/flask/app.py", line 902, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/src/emhass/web_server.py", line 414, in action_call
    input_data_dict = set_input_data_dict(
                      ^^^^^^^^^^^^^^^^^^^^
  File "/app/src/emhass/command_line.py", line 76, in set_input_data_dict
    params, retrieve_hass_conf, optim_conf, plant_conf = utils.treat_runtimeparams(
                                                         ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/src/emhass/utils.py", line 412, in treat_runtimeparams
    "delta_forecast_daily", runtimeparams["delta_forecast"]
                            ~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^
KeyError: 'delta_forecast'`

**To Reproduce**
Steps to reproduce the behavior

**Expected behavior**
No error

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Home Assistant installation type**
 - Home Assistant OS

**Your hardware**
- OS: HA OS
- Architecture: amd64

**EMHASS installation type**
 - Add-on, already with version 0.12.7 (I don't know about previous versions)

davidusb-geek, 2025-02-28:
This seems like a bug, I will look into it

g1za, 2025-03-01:
I think it may be somehow linked to the config json. I was testing the forecast on several days and even after updating the runtime param name in the command to `delta_forecast` it was still showing the results of just one day. Then I changed the value in the json as well and I got the 3 days I was looking for.
I hope it helps.

davidusb-geek, 2025-03-01:
Can you please explain what json are referring to and how and where did you change a value in a json?

g1za, 2025-03-01:
The EMHASS configuration. The one you can change from the GUI. Where you can set the cost function and so on.

davidusb-geek, 2025-03-01:
Ok so what it does not work is passing this at runtime?

g1za, 2025-03-01:
I think a mix. Not passing at runtime but also the name of the runtime param.
I changed the runtime name to `delta_forecast` and this suppressed the error in the log but I had the feeling the value was still not passed; the one from config was taken instead.

davidusb-geek, 2025-03-01:
Ok, for me this is solved with #476

g1za, 2025-03-01:
ok thank you!

closed 2025-03-01 by g1za

paulhomes, 2025-05-10:
Hi @davidusb-geek, perhaps re-open this issue as I don't think it has been resolved.  I am trying to use multi-day forecasts and also added `delta_forecast_daily` into my dayahead-optim REST API call and got the same error as @g1za with the latest version 0.13.3

```
NFO - emhass.web_server -  >> Obtaining params: 
INFO - emhass.web_server - Passed runtime parameters: {'delta_forecast_daily': 3, 'optimization_time_step': 30, 'set_use_adjusted_pv': False, 'publish_prefix': 'test_dau_'} 
INFO - emhass.web_server -  >> Setting input data dict 
INFO - emhass.web_server - Setting up needed data
ERROR - emhass.web_server - Exception on /action/dayahead-optim [POST]
Traceback (most recent call last):
  File "/app/.venv/lib/python3.12/site-packages/flask/app.py", line 1511, in wsgi_app
    response = self.full_dispatch_request()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/.venv/lib/python3.12/site-packages/flask/app.py", line 919, in full_dispatch_request
    rv = self.handle_user_exception(e)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/.venv/lib/python3.12/site-packages/flask/app.py", line 917, in full_dispatch_request
    rv = self.dispatch_request()
         ^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/.venv/lib/python3.12/site-packages/flask/app.py", line 902, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/src/emhass/web_server.py", line 411, in action_call
    input_data_dict = set_input_data_dict(
                      ^^^^^^^^^^^^^^^^^^^^
  File "/app/src/emhass/command_line.py", line 200, in set_input_data_dict
    params, retrieve_hass_conf, optim_conf, plant_conf = utils.treat_runtimeparams(
                                                         ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/src/emhass/utils.py", line 414, in treat_runtimeparams
    "delta_forecast_daily", runtimeparams["delta_forecast"]
                            ~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^
KeyError: 'delta_forecast'

```

If I also change it to  `delta_forecast` it works as expected.

The issue appears to be with these lines:

https://github.com/davidusb-geek/emhass/blob/9aff4456347b191ce8d9679ecb4fd16045bcf082/src/emhass/utils.py#L408-L419

It looks like it might try to use runtimeparams["delta_forecast"] as a default (for backward compatibility?) but fails because it is not present.  I encountered this whilst working on getting more than 3 days of open-meteo forecast data so will try to include a fix for this in that PR.