davidusb-geek/emhass #480: open-meteo negative PV forecast

g1za, 2025-03-02:
**Describe the bug**
I'm testing the new open-meteo PV forecast. In particular I'm noticing that if compared to my current metod based on forecast.solar, this always results in an infeasible optimization. Even if the profile of the PV forecast, load, costs and all the rest are exactly the same.
What I noticed and I'm wondering if may be the cause of this behaviour, is that I get some odd values of PV forecast at night. I see small negative values at night where I actually should see 0 and a tiny PV production for the current hour.
  

**To Reproduce**
Run the open-meteo forecast method

**Expected behavior**
No PV production at night, for sure no negative values.

**Screenshots**
Comparing here the charts from open-meteo and forecast.solar:
[image]
[image]

Here the table with the computation using open-meteo. As you can see there are negative values at night. I guess this, as part of the passed values, is not affected by the outcome of the optimization.

[image]
[image]

**Home Assistant installation type**
 - Home Assistant OS

**Your hardware**
- OS: HA OS
- Architecture: amd64

**EMHASS installation type**
 - Add-on

davidusb-geek, 2025-03-03:
Those negative values are certainly odd, I'll check that. But for me they are not likely to root cause for the infeasible outcome.
For now I cannot explain it.
Would it be possible to provide the payloads that you are passing in for optimal and infeasible cases?

purcell-lab, 2025-04-05:
With version 0.13 I am no longer seeings negative values from PV forecasts overnight.

g1za, 2025-04-05:
Correct, it's been fixed

closed 2025-04-05 by g1za