davidusb-geek/emhass #489: Error with v0.12.8 --> raise KeyError(f"{not_found} not in index")

Argogma, 2025-03-14:
After updating to v0.12.8 (before 0.12.6), I get an error message in the logs and the forecast is not working anymore. With the Backup to v0.12.6 the error is gone, the passed runtime parameters stay the same. Do you need more information?
```

[2025-03-14 16:00:00 +0100] [22] [INFO] Passed runtime parameters: {'pv_power_forecast': [1649, 377, 8, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 21, 535, 2007, 3847, 5504, 6316, 6254, 5621, 4613, 3530, 1974, 515, 12, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 50, 1838, 4381, 7186, 9568, 11733, 12411, 11458, 9208, 6234], 'alpha': 1, 'beta': 0, 'operating_hours_of_each_deferrable_load': [48, 5, 0, 0], 'prediction_horizon': 48, 'soc_init': 0.1, 'soc_final': 0.2}
[2025-03-14 16:00:00 +0100] [22] [INFO]  >> Setting input data dict
[2025-03-14 16:00:00 +0100] [22] [INFO] Setting up needed data
[2025-03-14 16:00:00 +0100] [22] [INFO] Retrieve hass get data method initiated...
[2025-03-14 16:00:03 +0100] [22] [ERROR] Exception on /action/naive-mpc-optim [POST]
Traceback (most recent call last):
  File "/app/.venv/lib/python3.12/site-packages/flask/app.py", line 1511, in wsgi_app
    response = self.full_dispatch_request()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/.venv/lib/python3.12/site-packages/flask/app.py", line 919, in full_dispatch_request
    rv = self.handle_user_exception(e)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/.venv/lib/python3.12/site-packages/flask/app.py", line 917, in full_dispatch_request
    rv = self.dispatch_request()
         ^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/.venv/lib/python3.12/site-packages/flask/app.py", line 902, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/src/emhass/web_server.py", line 414, in action_call
    input_data_dict = set_input_data_dict(
                      ^^^^^^^^^^^^^^^^^^^^
  File "/app/src/emhass/command_line.py", line 284, in set_input_data_dict
    if not rh.prepare_data(
           ^^^^^^^^^^^^^^^^
  File "/app/src/emhass/retrieve_hass.py", line 397, in prepare_data
    self.df_final[new_var_replace_zero] = self.df_final[
                                          ^^^^^^^^^^^^^^
  File "/app/.venv/lib/python3.12/site-packages/pandas/core/frame.py", line 4108, in __getitem__
    indexer = self.columns._get_indexer_strict(key, "columns")[1]
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/.venv/lib/python3.12/site-packages/pandas/core/indexes/base.py", line 6200, in _get_indexer_strict
    self._raise_if_missing(keyarr, indexer, axis_name)
  File "/app/.venv/lib/python3.12/site-packages/pandas/core/indexes/base.py", line 6252, in _raise_if_missing
    raise KeyError(f"{not_found} not in index")
KeyError: "['sensor.inverter_power_kombiniert'] not in index"
[2025-03-14 16:00:03 +0100] [22] [INFO]  >> Obtaining params: 
[2025-03-14 16:00:03 +0100] [22] [INFO] Passed runtime parameters: {}
[2025-03-14 16:00:03 +0100] [22] [INFO]  >> Setting input data dict
[2025-03-14 16:00:03 +0100] [22] [INFO] Setting up needed data
[2025-03-14 16:00:03 +0100] [22] [INFO]  >> Publishing data...
[2025-03-14 16:00:03 +0100] [22] [INFO] Publishing data to HASS instance
[2025-03-14 16:00:03 +0100] [22] [INFO] Successfully posted to sensor.p_pv_forecast = 1663
[2025-03-14 16:00:03 +0100] [22] [INFO] Successfully posted to sensor.p_load_forecast = 299.93
[2025-03-14 16:00:03 +0100] [22] [INFO] Successfully posted to sensor.p_deferrable0 = 2500.0
[2025-03-14 16:00:03 +0100] [22] [INFO] Successfully posted to sensor.p_deferrable1 = 0.0
[2025-03-14 16:00:03 +0100] [22] [INFO] Successfully posted to sensor.p_deferrable2 = 0.0
[2025-03-14 16:00:03 +0100] [22] [INFO] Successfully posted to sensor.p_deferrable3 = 0.0
[2025-03-14 16:00:03 +0100] [22] [INFO] Successfully posted to sensor.p_batt_forecast = 0.0
[2025-03-14 16:00:03 +0100] [22] [INFO] Successfully posted to sensor.soc_batt_forecast = 11.0
[2025-03-14 16:00:03 +0100] [22] [INFO] Successfully posted to sensor.p_grid_forecast = 1136.93
[2025-03-14 16:00:03 +0100] [22] [INFO] Successfully posted to sensor.total_cost_fun_value = -20.34
[2025-03-14 16:00:03 +0100] [22] [INFO] Successfully posted to sensor.optim_status = Optimal
[2025-03-14 16:00:03 +0100] [22] [INFO] Successfully posted to sensor.unit_load_cost = 0.2467
[2025-03-14 16:00:03 +0100] [22] [INFO] Successfully posted to sensor.unit_prod_price = 0.078

```
__________________________________________

My Config:

```
{
"battery_charge_efficiency": 0.85,
"battery_charge_power_max": 5000,
"battery_discharge_efficiency": 0.85,
"battery_discharge_power_max": 5000,
"battery_dynamic_max": 0.9,
"battery_dynamic_min": -0.9,
"battery_maximum_state_of_charge": 1,
"battery_minimum_state_of_charge": 0.05,
"battery_nominal_energy_capacity": 10000,
"battery_target_state_of_charge": 0.5,
"compute_curtailment": false,
"continual_publish": true,
"costfun": "cost",
"delta_forecast_daily": 2,
"end_timesteps_of_each_deferrable_load": [
  0,
  0,
  0,
  5
],
"historic_days_to_retrieve": 14,
"inverter_is_hybrid": false,
"load_cost_forecast_method": "hp_hc_periods",
"load_forecast_method": "naive",
"load_negative": false,
"load_offpeak_hours_cost": 0.2467,
"load_peak_hour_periods": {
  "period_hp_1": [
    {
      "start": "00:01"
    },
    {
      "end": "23:59"
    }
  ]
},
"load_peak_hours_cost": 0.2467,
"logging_level": "DEBUG",
"lp_solver": "default",
"lp_solver_path": "empty",
"maximum_power_from_grid": 15000,
"maximum_power_to_grid": 15000,
"method_ts_round": "nearest",
"modules_per_string": [
  16
],
"nominal_power_of_deferrable_loads": [
  2500,
  670,
  230,
  7500
],
"number_of_deferrable_loads": 4,
"operating_hours_of_each_deferrable_load": [
  24,
  6,
  2,
  2
],
"optimization_time_step": 60,
"photovoltaic_production_sell_price": 0.078,
"production_price_forecast_method": "constant",
"pv_inverter_model": [
  "Fronius_International_GmbH__Fronius_Primo_5_0_1_208_240__240V_"
],
"pv_module_model": [
  "CSUN_Eurasia_Energy_Systems_Industry_and_Trade_CSUN295_60M"
],
"sensor_linear_interp": [
  "sensor.inverter_power_kombiniert",
  "sensor.power_load_no_var_loads"
],
"sensor_power_load_no_var_loads": "sensor.power_load_no_var_loads",
"sensor_power_photovoltaics": "sensor.inverter_power_kombiniert",
"sensor_replace_zero": [
  "sensor.inverter_power_kombiniert",
  "sensor.power_load_no_var_loads"
],
"set_battery_dynamic": false,
"set_deferrable_load_single_constant": [
  false,
  false,
  false,
  true
],
"set_deferrable_startup_penalty": [
  0,
  0,
  0,
  0
],
"set_nocharge_from_grid": false,
"set_nodischarge_to_grid": true,
"set_total_pv_sell": false,
"set_use_battery": true,
"set_use_pv": false,
"set_zero_min": true,
"start_timesteps_of_each_deferrable_load": [
  0,
  0,
  0,
  0
],
"strings_per_inverter": [
  1
],
"surface_azimuth": [
  205
],
"surface_tilt": [
  30
],
"treat_deferrable_load_as_semi_cont": [
  false,
  false,
  false,
  true
],
"weather_forecast_method": "open-meteo",
"weight_battery_charge": 0,
"weight_battery_discharge": 0
}

```
**EMHASS installation type**
 - Add-on

**Additional context**
Add any other context about the problem here.

MarcoPolet, 2025-03-15:
I have exactly the same issue, when upgrading from 0.12.5 to 0.12.8. My message concerns the sensor solarnet_power_photovoltaics
Seems like it's also the solar power sensor like yours. The sensor is one of two in the config item 'Sensor to replace NAN values with linear interpolation:'

MarcoPolet, 2025-03-15:
One thing I noticed as difference between 0.12.5 and 0.12.8 is at startup the data path is different (/app/data and /data). I modified my docker to add a volume to /data, but that did not solve the issue. Also cleaning this folder and running a new day-ahead did not resolve anything.
Note: the day-ahead call is working fine

tbrasser, 2025-03-15:
Same here, but 0.12.7 works great, 0.12.8 mpc doesn't.

davidusb-geek, 2025-03-16:
Not 100% sure but I think that this issue will be solved in the upcoming release. I just need some more time to finish the developments for the new release

MarcoPolet, 2025-04-06:
Hi @davidusb-geek , I tried the 0.13.0 yesterday, but unfortunately the issue from above is not solved in this version. No pressure, I'm happy using 0.12.7, but just to let you know.

davidusb-geek, 2025-04-21:
This should be fixed now.
Re-open if it's not the case.

Argogma, 2025-04-21:
I tried the v0.13.1 (before 0.12.6) and get the following error, which I don't know if it is the same source of error?

[2025-04-21 23:02:04 +0200] [21] [ERROR] Exception on /action/naive-mpc-optim [POST]
Traceback (most recent call last):
  File "/app/.venv/lib/python3.12/site-packages/pandas/core/indexes/base.py", line 3805, in get_loc
    return self._engine.get_loc(casted_key)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "index.pyx", line 167, in pandas._libs.index.IndexEngine.get_loc
  File "index.pyx", line 196, in pandas._libs.index.IndexEngine.get_loc
  File "pandas/_libs/hashtable_class_helper.pxi", line 7081, in pandas._libs.hashtable.PyObjectHashTable.get_item
  File "pandas/_libs/hashtable_class_helper.pxi", line 7089, in pandas._libs.hashtable.PyObjectHashTable.get_item
KeyError: 'sensor.inverter_power_kombiniert'
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/app/.venv/lib/python3.12/site-packages/flask/app.py", line 1511, in wsgi_app
    response = self.full_dispatch_request()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/.venv/lib/python3.12/site-packages/flask/app.py", line 919, in full_dispatch_request
    rv = self.handle_user_exception(e)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/.venv/lib/python3.12/site-packages/flask/app.py", line 917, in full_dispatch_request
    rv = self.dispatch_request()
         ^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/.venv/lib/python3.12/site-packages/flask/app.py", line 902, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/src/emhass/web_server.py", line 411, in action_call
    input_data_dict = set_input_data_dict(
                      ^^^^^^^^^^^^^^^^^^^^
  File "/app/src/emhass/command_line.py", line 382, in set_input_data_dict
    P_PV_forecast = fcst.get_power_from_weather(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/src/emhass/forecast.py", line 790, in get_power_from_weather
    P_PV_forecast = Forecast.get_mix_forecast(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/src/emhass/forecast.py", line 690, in get_mix_forecast
    first_fcst = alpha * df_forecast.iloc[0] + beta * df_now[col].iloc[-1]
                                                      ~~~~~~^^^^^
  File "/app/.venv/lib/python3.12/site-packages/pandas/core/frame.py", line 4102, in __getitem__
    indexer = self.columns.get_loc(key)
              ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/.venv/lib/python3.12/site-packages/pandas/core/indexes/base.py", line 3812, in get_loc
    raise KeyError(key) from err
KeyError: 'sensor.inverter_power_kombiniert'

paulhomes, 2025-04-21:
I worked on some code around this area in 0.13.1 and would like to see if I can help. Please can you turn on debug level logging and post the relevant section of the log. I am particularly interesting in the log section between _[INFO]  >> Obtaining params:_ and _[DEBUG] get_weather_forecast returning:_

MarcoPolet, 2025-04-22:
Hi @paulhomes 
I have the same as described by @Argogma above. I included some more debuglogs. A few things seem strange:
`[2025-04-22 15:45:02 +0200] [23] [DEBUG] get_weather_forecast returning:

                           yhat

ts                             

2025-04-22 15:45:00+02:00  5822`

Which almost seems like a memory-buffer-overrun or something.

Another thing I noticed:
`[2025-04-22 15:45:16 +0200] [23] [WARNING] Sensor 'sensor.deye_total_solar_power' in var_interp not found in self.var_list and has been removed.
`
This is probably causing the follow-up error @Argogma (and I) have. But I have no idea what the warning means.

Thanks for your help, let me know if you need more input.

Regards,
Marco

`⠋ Preparing packages... (0/0)                                                   
⠋ Preparing packages... (0/1)                                                   
⠙ Preparing packages... (0/1)                                                   
   Building emhass @ file:///app

⠙ Preparing packages... (0/1)                                                   

   Building emhass @ file:///app

⠹ Preparing packages... (0/1)                                                   

   Building emhass @ file:///app

⠸ Preparing packages... (0/1)                                                   

   Building emhass @ file:///app

⠼ Preparing packages... (0/1)                                                   

   Building emhass @ file:///app

⠴ Preparing packages... (0/1)                                                   

   Building emhass @ file:///app

⠦ Preparing packages... (0/1)                                                   

   Building emhass @ file:///app

⠧ Preparing packages... (0/1)                                                   

   Building emhass @ file:///app

⠇ Preparing packages... (0/1)                                                   

   Building emhass @ file:///app

⠋ Preparing packages... (0/1)                                                   

   Building emhass @ file:///app

⠙ Preparing packages... (0/1)                                                   

   Building emhass @ file:///app

⠹ Preparing packages... (0/1)                                                   

      Built emhass @ file:///app
[... 530 lines cut ...]

  File "/app/src/emhass/forecast.py", line 790, in get_power_from_weather

    P_PV_forecast = Forecast.get_mix_forecast(

                    ^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "/app/src/emhass/forecast.py", line 690, in get_mix_forecast

    first_fcst = alpha * df_forecast.iloc[0] + beta * df_now[col].iloc[-1]

                                                      ~~~~~~^^^^^

  File "/app/.venv/lib/python3.12/site-packages/pandas/core/frame.py", line 4102, in __getitem__

    indexer = self.columns.get_loc(key)

              ^^^^^^^^^^^^^^^^^^^^^^^^^

  File "/app/.venv/lib/python3.12/site-packages/pandas/core/indexes/base.py", line 3812, in get_loc

    raise KeyError(key) from err

KeyError: 'sensor.deye_total_solar_power'
`

MarcoPolet, 2025-04-22:
Note in my log above there are 2 day-ahead tries, first one also has an issue with not passing enough elements, second one (from around 16:00) does not have that.

paulhomes, 2025-04-22:
Hi @Argogma , please can you re-open this issue as I don't have permission to do so myself.

paulhomes, 2025-04-22:
Hi @MarcoPolet , there was some code fixed in 0.13.1 that originally intended to remove extraneous items specified in config.json `sensor_replace_zero` and `sensor_linear_interp` that were not found in the variable list. Now that has changed, it seems to be highlighting other issues. 

Thanks for posting the debug log. I can see that for some reason the variable list for you only contains `sensor.power_load_no_var_loads` and not both `sensor.deye_total_solar_power` and `sensor.power_load_no_var_loads` as specified in `sensor_linear_interp`. As `sensor.deye_total_solar_power` is used later I would have expected it to be present in the variable list. I'd like to investigate that some more and see if I can replicate the issue.

Please can you post your config.json (either as an attachment or in a triple back-ticks code block to preserve formatting) so we can see what is specified there.

Argogma, 2025-04-22:
Hmmm, i dont have permission  to open again ?!?
Tomorrow evening i can post my logs if needed :)

paulhomes, 2025-04-22:
@davidusb-geek please can you re-open this issue as we don't have permission to. Thanks

MarcoPolet, 2025-04-24:
Hi @paulhomes , thank you so much for looking into this!
Here's my config:
```{
    "logging_level": "DEBUG",
    "costfun": "profit",
    "optimization_time_step": 15,
    "historic_days_to_retrieve": 9,
    "method_ts_round": "first",
    "continual_publish": false,
    "data_path": "default",
    "set_total_pv_sell": false,
    "lp_solver": "default",
    "lp_solver_path": "empty",
    "set_nocharge_from_grid": false,
    "set_nodischarge_to_grid": false,
    "set_battery_dynamic": false,
    "battery_dynamic_max": 0.9,
    "battery_dynamic_min": -0.9,
    "weight_battery_discharge": 1,
    "weight_battery_charge": 1,
    "sensor_power_photovoltaics": "sensor.deye_total_solar_power",
    "sensor_power_photovoltaics_forecast": "sensor.p_pv_forecast",
    "sensor_power_load_no_var_loads": "sensor.power_load_no_var_loads",
    "sensor_replace_zero": [],
    "sensor_linear_interp": [
        "sensor.deye_total_solar_power",
        "sensor.power_load_no_var_loads"
    ],
    "load_negative": false,
    "set_zero_min": true,
    "number_of_deferrable_loads": 3,
    "nominal_power_of_deferrable_loads": [
        6600,
        3000,
        1000
    ],
    "minimum_power_of_deferrable_loads": [
        0,
        0
    ],
    "operating_hours_of_each_deferrable_load": [
        0,
        0,
        0
    ],
    "weather_forecast_method": "solcast",
    "load_forecast_method": "naive",
    "delta_forecast_daily": 1,
    "load_cost_forecast_method": "hp_hc_periods",
    "start_timesteps_of_each_deferrable_load": [
        0,
        0,
        0
    ],
    "end_timesteps_of_each_deferrable_load": [
        0,
        0,
        0
    ],
    "load_peak_hour_periods": {
        "period_hp_1": [
            {
                "start": "02:54"
            },
            {
                "end": "15:24"
            }
        ],
        "period_hp_2": [
            {
                "start": "17:24"
            },
            {
                "end": "20:24"
            }
        ]
    },
    "treat_deferrable_load_as_semi_cont": [
        true,
        true,
        true
    ],
    "set_deferrable_load_single_constant": [
        false,
        false,
        false
    ],
    "set_deferrable_startup_penalty": [
        0,
        0,
        0
    ],
    "load_peak_hours_cost": 0.1907,
    "load_offpeak_hours_cost": 0.1419,
    "production_price_forecast_method": "constant",
    "photovoltaic_production_sell_price": 0.1419,
    "maximum_power_from_grid": 17250,
    "maximum_power_to_grid": 17250,
    "set_use_pv": false,
    "set_use_adjusted_pv": false,
    "adjusted_pv_regression_model": "LassoRegression",
    "adjusted_pv_solar_elevation_threshold": 10,
    "pv_module_model": [
        "CSUN_Eurasia_Energy_Systems_Industry_and_Trade_CSUN295_60M"
    ],
    "pv_inverter_model": [
        "Fronius_International_GmbH__Fronius_Primo_5_0_1_208_240__240V_"
    ],
    "surface_tilt": [
        30
    ],
    "surface_azimuth": [
        205
    ],
    "modules_per_string": [
        16
    ],
    "strings_per_inverter": [
        1
    ],
    "inverter_is_hybrid": false,
    "compute_curtailment": false,
    "set_use_battery": true,
    "battery_discharge_power_max": 7680,
    "battery_charge_power_max": 7680,
    "battery_discharge_efficiency": 0.95,
    "battery_charge_efficiency": 0.95,
    "battery_nominal_energy_capacity": 9600,
    "battery_minimum_state_of_charge": 0.05,
    "battery_maximum_state_of_charge": 0.95,
    "battery_target_state_of_charge": 0.1
}
```
Fyi, there is info in the config regarding my solar panels configuration, but I override that in the curl command. Same for the solar forecast & kWh prices, they are passed by a list (as shown in my log above).

And thanks for the three-ticks way of working, did not know that :-)

Argogma, 2025-04-24:
Hey @paulhomes,

here is my config and once more the protocoll :)
Do you need something else?

```
{
  "adjusted_pv_regression_model": "LassoRegression",
  "adjusted_pv_solar_elevation_threshold": 10,
  "battery_charge_efficiency": 0.85,
  "battery_charge_power_max": 5000,
  "battery_discharge_efficiency": 0.85,
  "battery_discharge_power_max": 5000,
  "battery_dynamic_max": 0.9,
  "battery_dynamic_min": -0.9,
  "battery_maximum_state_of_charge": 1,
  "battery_minimum_state_of_charge": 0.05,
  "battery_nominal_energy_capacity": 10000,
  "battery_target_state_of_charge": 0.5,
  "compute_curtailment": false,
  "continual_publish": true,
  "costfun": "cost",
  "delta_forecast_daily": 2,
  "end_timesteps_of_each_deferrable_load": [
    0,
    0,
    0,
    5
  ],
  "historic_days_to_retrieve": 14,
  "inverter_is_hybrid": false,
  "load_cost_forecast_method": "hp_hc_periods",
  "load_forecast_method": "naive",
  "load_negative": false,
  "load_offpeak_hours_cost": 0.2467,
  "load_peak_hour_periods": {
    "period_hp_1": [
      {
        "start": "00:01"
      },
      {
        "end": "23:59"
      }
    ]
  },
  "load_peak_hours_cost": 0.2467,
  "logging_level": "DEBUG",
  "lp_solver": "default",
  "lp_solver_path": "empty",
  "lp_solver_timeout": 45,
  "maximum_power_from_grid": 15000,
  "maximum_power_to_grid": 15000,
  "method_ts_round": "nearest",
  "minimum_power_of_deferrable_loads": [
    0,
    0
  ],
  "modules_per_string": [
    16
  ],
  "nominal_power_of_deferrable_loads": [
    2500,
    670,
    230,
    7500
  ],
  "number_of_deferrable_loads": 4,
  "open_meteo_cache_max_age": 30,
  "operating_hours_of_each_deferrable_load": [
    24,
[... 297 lines cut ...]
         ^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/.venv/lib/python3.12/site-packages/flask/app.py", line 902, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/src/emhass/web_server.py", line 411, in action_call
    input_data_dict = set_input_data_dict(
                      ^^^^^^^^^^^^^^^^^^^^
  File "/app/src/emhass/command_line.py", line 382, in set_input_data_dict
    P_PV_forecast = fcst.get_power_from_weather(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/src/emhass/forecast.py", line 790, in get_power_from_weather
    P_PV_forecast = Forecast.get_mix_forecast(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/src/emhass/forecast.py", line 690, in get_mix_forecast
    first_fcst = alpha * df_forecast.iloc[0] + beta * df_now[col].iloc[-1]
                                                      ~~~~~~^^^^^
  File "/app/.venv/lib/python3.12/site-packages/pandas/core/frame.py", line 4102, in __getitem__
    indexer = self.columns.get_loc(key)
              ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/.venv/lib/python3.12/site-packages/pandas/core/indexes/base.py", line 3812, in get_loc
    raise KeyError(key) from err
KeyError: 'sensor.inverter_power_kombiniert'
[2025-04-24 08:45:02 +0200] [22] [INFO]  >> Obtaining params: 
[2025-04-24 08:45:02 +0200] [22] [INFO] Passed runtime parameters: {}
```

davidusb-geek, 2025-04-24:
> Although you have set_use_pv as false in your config, from my reading of the code I think it is ignoring that and still trying to do PV processing because you passed in a weather forecast list in the `pv_power_forecast` parameter

This is correct.
If we receive a passed weather forecast then we process it and use it. 
But maybe the bug is that then on that case we are not setting set_use_pv to True afterwards.

paulhomes, 2025-04-24:
Hi @MarcoPolet , I think it may be the same issue for you too as you also have `"set_use_pv": false` in your config.json. If you change it to true does it resolve the issue for you?

Argogma, 2025-04-24:
Perfect!!! That seems to solve the problem for me!

> Hi [@Argogma](https://github.com/Argogma) , thanks for providing that detailed info. I think the core of the issue is that you have `"set_use_pv": false` in your config.json and the `"sensor_power_photovoltaics": "sensor.inverter_power_kombiniert"` config is only added to the list of sensors to fetch from Home Assistant when set_use_pv is true. There were also code changes recently that removes sensors from the `sensor_replace_zero` and `sensor_linear_interp` lists when they are not present in that variable list which is the reason for the new warning messages.
> 
> If you change set_use_pv from false to true in your config.json does it resolve the issue for you?
> 
> Although you have set_use_pv as false in your config, from my reading of the code I think it is ignoring that and still trying to do PV processing because you passed in a weather forecast list in the `pv_power_forecast` parameter. I think there may be a bug here in that perhaps set_use_pv should be an implied true if pv_power_forecast is passed to avoid the ambiguity. What do you think [@davidusb-geek](https://github.com/davidusb-geek) ? The code sections I'm referring to are
> 
> [emhass/src/emhass/command_line.py](https://github.com/davidusb-geek/emhass/blob/55be2aab4509dc8d5e2d738744cce3a6b0392348/src/emhass/command_line.py#L72-L73)
> 
> Lines 72 to 73 in [55be2aa](/davidusb-geek/emhass/commit/55be2aab4509dc8d5e2d738744cce3a6b0392348)
>  if optim_conf.get("set_use_pv", True): 
>      var_list.append(retrieve_hass_conf["sensor_power_photovoltaics"]) 
> and
> 
> [emhass/src/emhass/command_line.py](https://github.com/davidusb-geek/emhass/blob/55be2aab4509dc8d5e2d738744cce3a6b0392348/src/emhass/command_line.py#L277-L280)
> 
> Lines 277 to 280 in [55be2aa](/davidusb-geek/emhass/commit/55be2aab4509dc8d5e2d738744cce3a6b0392348)
>  if ( 
>      optim_conf["set_use_pv"] 
>      or optim_conf.get("weather_forecast_method", None) == "list" 
>  ):

closed 2025-04-24 by Argogma

MarcoPolet, 2025-04-25:
@paulhomes Yes! Turning the use-pv on solved it! One question though: I'm providing the forecast (from solcast) through a list at the curl commandline, will this be used instead of the (fake) pv-inverter I setup to enable the use-pv button?

davidusb-geek, 2025-04-26:
@MarcoPolet Yes if you provide your own forecast on the curl then that is what will be used. The inverter and panel settings are only used with the open-meteo method