davidusb-geek/emhass #492: Feature request: refresh stale weather cache when required rather than error and wait for next iteration

paulhomes, 2025-03-17:
If I understand this correctly, it looks like when using /action/naive-mpc-optim with weather_forecast_cache: true, and the weather cache is stale, it logs an error/warning and returns HTTP 400. The next call will then refreshes the weather cache and continues as normal.

What do you think of refreshing the weather cache immediately on the first call (assuming weather_forecast_cache: true is passed) instead of logging an error?

In the end it all works out ok as-is, but refreshing automatically when required might help keep the logs clear of errors and avoid subsequent investigations and questions.

This is a docker log showing the initial cache error at 13:15 and subsequent fetch/cache at 13:20 
```
emhass  | [2025-03-17 13:15:30 +1000] [25] [INFO]  >> Obtaining params: 
emhass  | [2025-03-17 13:15:30 +1000] [25] [INFO] Passed runtime parameters: {'load_cost_forecast': [0.06, 0.06, 0.13, 0.15, 0.13, 0.1, 0.28, 0.33, 0.34, 0.45, 0.45, 0.43, 0.37, 0.34, 0.33, 0.31, 0.22, 0.21, 0.23, 0.22, 0.21, 0.2, 0.2, 0.2, 0.2, 0.2, 0.2, 0.2, 0.2, 0.2, 0.2, 0.2, 0.23, 0.23, 0.24, 0.24, 0.23, 0.2, 0.18, 0.18, 0.15, 0.13, 0.13, 0.1, 0.09, 0.07, 0.06, 0.07], 'prod_price_forecast': [-0.03, -0.02, 0.04, 0.06, 0.04, 0.01, 0.04, 0.08, 0.09, 0.19, 0.19, 0.18, 0.12, 0.09, 0.08, 0.07, 0.09, 0.09, 0.1, 0.1, 0.09, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.1, 0.11, 0.12, 0.11, 0.1, 0.08, 0.06, 0.06, 0.06, 0.04, 0.04, 0.01, -0.0, -0.02, -0.02, -0.02], 'prediction_horizon': 48, 'soc_init': 0.98, 'soc_final': 0.98, 'weather_forecast_cache': True}
emhass  | [2025-03-17 13:15:30 +1000] [25] [INFO]  >> Setting input data dict
emhass  | [2025-03-17 13:15:30 +1000] [25] [INFO] Setting up needed data
emhass  | [2025-03-17 13:15:30 +1000] [25] [INFO] Retrieve hass get data method initiated...
emhass  | [2025-03-17 13:15:30 +1000] [25] [INFO] Retrieving weather forecast data using method = open-meteo
emhass  | [2025-03-17 13:15:30 +1000] [25] [ERROR] Unable to obtain cached forecast data within the requested timeframe range.
emhass  | [2025-03-17 13:15:30 +1000] [25] [ERROR] Try running optimization again (not using cache). Optionally, add runtime parameter 'weather_forecast_cache': true to pull new data from forecast API and cache.
emhass  | [2025-03-17 13:15:30 +1000] [25] [WARNING] Removing old forecast cache file. Next optimization will pull data from forecast API, unless 'weather_forecast_cache_only': true
emhass  | [2025-03-17 13:20:30 +1000] [25] [INFO]  >> Obtaining params: 
emhass  | [2025-03-17 13:20:30 +1000] [25] [INFO] Passed runtime parameters: {'load_cost_forecast': [0.07, 0.09, 0.13, 0.15, 0.13, 0.1, 0.28, 0.33, 0.34, 0.45, 0.45, 0.43, 0.37, 0.34, 0.33, 0.31, 0.22, 0.21, 0.23, 0.22, 0.21, 0.2, 0.2, 0.2, 0.2, 0.2, 0.2, 0.2, 0.2, 0.2, 0.2, 0.2, 0.23, 0.23, 0.24, 0.24, 0.23, 0.2, 0.18, 0.18, 0.15, 0.13, 0.13, 0.1, 0.09, 0.07, 0.06, 0.07], 'prod_price_forecast': [-0.01, -0.0, 0.04, 0.06, 0.04, 0.01, 0.04, 0.08, 0.09, 0.19, 0.19, 0.18, 0.12, 0.09, 0.08, 0.07, 0.09, 0.09, 0.1, 0.1, 0.09, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.1, 0.11, 0.12, 0.11, 0.1, 0.08, 0.06, 0.06, 0.06, 0.04, 0.04, 0.01, -0.0, -0.02, -0.02, -0.02], 'prediction_horizon': 48, 'soc_init': 0.98, 'soc_final': 0.98, 'weather_forecast_cache': True}
emhass  | [2025-03-17 13:20:30 +1000] [25] [INFO]  >> Setting input data dict
emhass  | [2025-03-17 13:20:30 +1000] [25] [INFO] Setting up needed data
emhass  | [2025-03-17 13:20:30 +1000] [25] [INFO] Retrieve hass get data method initiated...
emhass  | [2025-03-17 13:20:30 +1000] [25] [INFO] Retrieving weather forecast data using method = open-meteo
emhass  | [2025-03-17 13:20:32 +1000] [25] [INFO] Saved the forecast results to cache, for later reference.
emhass  | [2025-03-17 13:20:32 +1000] [25] [INFO] Retrieving data from hass for load forecast using method = naive
emhass  | [2025-03-17 13:20:32 +1000] [25] [INFO] Retrieve hass get data method initiated...
emhass  | [2025-03-17 13:20:34 +1000] [25] [INFO]  >> Performing naive MPC optimization...
emhass  | [2025-03-17 13:20:34 +1000] [25] [INFO] Performing naive MPC optimization
emhass  | [2025-03-17 13:20:34 +1000] [25] [INFO] Perform an iteration of a naive MPC controller
emhass  | [2025-03-17 13:20:34 +1000] [25] [INFO] Status: Optimal
emhass  | [2025-03-17 13:20:34 +1000] [25] [INFO] Total value of the Cost function = -0.29
emhass  | [2025-03-17 13:20:34 +1000] [25] [INFO]  >> Obtaining params: 
emhass  | [2025-03-17 13:20:34 +1000] [25] [INFO] Passed runtime parameters: {}
emhass  | [2025-03-17 13:20:34 +1000] [25] [INFO]  >> Setting input data dict
emhass  | [2025-03-17 13:20:34 +1000] [25] [INFO] Setting up needed data
emhass  | [2025-03-17 13:20:34 +1000] [25] [INFO]  >> Publishing data...
emhass  | [2025-03-17 13:20:34 +1000] [25] [INFO] Publishing data to HASS instance
emhass  | [2025-03-17 13:20:34 +1000] [25] [INFO] Successfully posted to sensor.p_pv_forecast = 7791.94
emhass  | [2025-03-17 13:20:34 +1000] [25] [INFO] Successfully posted to sensor.p_load_forecast = 2845.9
emhass  | [2025-03-17 13:20:34 +1000] [25] [INFO] Successfully posted to sensor.p_pv_curtailment = 4496.49
emhass  | [2025-03-17 13:20:34 +1000] [25] [INFO] Successfully posted to sensor.p_hybrid_inverter = 2845.9
emhass  | [2025-03-17 13:20:34 +1000] [25] [INFO] Successfully posted to sensor.p_batt_forecast = -449.55
emhass  | [2025-03-17 13:20:34 +1000] [25] [INFO] Successfully posted to sensor.soc_batt_forecast = 98.42
emhass  | [2025-03-17 13:20:34 +1000] [25] [INFO] Successfully posted to sensor.p_grid_forecast = 0.0
emhass  | [2025-03-17 13:20:34 +1000] [25] [INFO] Successfully posted to sensor.total_cost_fun_value = 1.9
emhass  | [2025-03-17 13:20:34 +1000] [25] [INFO] Successfully posted to sensor.optim_status = Optimal
emhass  | [2025-03-17 13:20:34 +1000] [25] [INFO] Successfully posted to sensor.unit_load_cost = 0.07
emhass  | [2025-03-17 13:20:34 +1000] [25] [INFO] Successfully posted to sensor.unit_prod_price = -0.01
```

paulhomes, 2025-03-17:
I've been looking further into the open-meteo weather caching and noticed that even though open-meteo is returning 72h of weather by default (emhass doesn't look like it passes a forecast_days parameter to change this) the 24h prediction_horizon I am passing into an mpc call (with weather_forecast_cache: true) means that emhass subsets and only caches the next 24h (and not the full 72h returned by open-meteo). This means that the cache has insufficient data within 15-30 minutes for future mpc calls and so open-meteo gets pinged regularly. What are your thoughts on saving/loading the cached open-meteo data before any emhass date filtering gets done? On subsequent mpc calls, date filtering can be done after loading the cached weather back into memory. By keeping a longer horizon in the cache the number of open-meteo API calls can be reduced.

I do note that this means the cache may then not get refreshed for another 48h unless there is another mechanism to empty it after a configurable amount of time - or via a specific action - as suggested in #420

davidusb-geek, 2025-04-21:
Closing as completed

closed 2025-04-21 by davidusb-geek