davidusb-geek/emhass #495: Missing files in python package after 0.12.5

pail23, 2025-03-25:
**Describe the bug**
There are files missing in the python package after version 0.12.4

**To Reproduce**
Steps to reproduce the behavior:
Run the command `uv add emhass==0.12.8`

then, I get the following result for `ls .venv/lib/python3.12/site-packages/emhass`

`data  static  templates
`

When I import `utils` from `emhass`:
`from emhass import utils `
I get an exception:
`ImportError: cannot import name 'utils' from 'emhass' (unknown location)`

**Expected behavior**

When I install the package with `uv add emhass==0.12.4`, I get the following result for `ls .venv/lib/python3.12/site-packages/emhass`:

```
__init__.py  __pycache__  command_line.py  data  forecast.py  img  machine_learning_forecaster.py  machine_learning_regressor.py  optimization.py  retrieve_hass.py  static  templates  utils.py  web_server.py
```

`from emhass import utils ` is working in version 0.12.4. This is the expected behaviour.

Maybe changing the sdist definition in pyproject.toml could fix the issue:

```
[tool.hatch.build.targets.sdist]
packages = ["src/emhass"]
```

**EMHASS installation type**
 - Legacy Python virtual environment (using emhass as library)

davidusb-geek, 2025-03-25:
Will look into it.
But it seems weird that we don't have these problems when building our containers as we are using similar commands

pail23, 2025-03-26:
Many thanks.

As a side note. You are actually not installing emhass in the container using the python package (uv add emhass). Instead you are coping it from the source. See `Dockerfile`:

```
# copy required EMHASS files
COPY src/emhass/ /app/src/emhass/

# webserver files
COPY src/emhass/templates/ /app/src/emhass/templates/
COPY src/emhass/static/ /app/src/emhass/static/
COPY src/emhass/static/data/ /app/src/emhass/static/data/
COPY src/emhass/static/img/ /app/src/emhass/static/img/
```

davidusb-geek, 2025-04-02:
Hi.
We will keep the install from source method on the Dockerfile.
Just added the attempt to fix this in PR #476 
But just as an additional comment the legacy installation method using `pip` the install from source method works just fine.
You could just do the following on the cloned code (inside a venv preferably):
```
pip install .
```

davidusb-geek, 2025-04-02:
The proposed change on pyproject.toml did not solved the issue.

davidusb-geek, 2025-04-21:
Hi @GeoDerp, I've just tested this and unfortunately the problem persists.

This problem and the issue raised in #495 is about the uv implementation.
But I would expect that we should be able to install and use emhass the old way.

For example my test is simply creating a new venv, then activating and installing the latests version of emhass.
So I'm doing:
```
python -m venv .venv
.venv\Scripts\activate
python -m pip install --uprade pip
pip install emhass==0.13.1
```

Then when trying to import:
```
python
from emhass import utils
```

Results in:

[image]

pail23, 2025-04-26:
I gave it another try with #527:

```
[tool.hatch.build.targets.sdist]
include = [
    "src/emhass/*.py",
    "src/emhass/templates/",
    "src/emhass/static/",
    "src/emhass/img/",
    "src/emhass/data/cec_modules.pbz2",
    "src/emhass/data/cec_inverters.pbz2",
    "src/emhass/data/associations.csv",
    "src/emhass/data/config_defaults.json",
]
```

pail23, 2025-04-29:
Solved in 0.13.2

closed 2025-04-29 by pail23