davidusb-geek/emhass #496: Daylight savings time, optimazations fail

llevering, 2025-03-28:
**Describe the bug**
Optimisations (day-ahead & MPC) fail 

The log contains:
[2025-03-28 22:00:17 +0100] [25] [ERROR] Exception on /action/naive-mpc-optim [POST]
Traceback (most recent call last):
  File "/app/.venv/lib/python3.12/site-packages/flask/app.py", line 1511, in wsgi_app
    response = self.full_dispatch_request()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/.venv/lib/python3.12/site-packages/flask/app.py", line 919, in full_dispatch_request
    rv = self.handle_user_exception(e)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/.venv/lib/python3.12/site-packages/flask/app.py", line 917, in full_dispatch_request
    rv = self.dispatch_request()
         ^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/.venv/lib/python3.12/site-packages/flask/app.py", line 902, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/src/emhass/web_server.py", line 414, in action_call
    input_data_dict = set_input_data_dict(
                      ^^^^^^^^^^^^^^^^^^^^
  File "/app/src/emhass/command_line.py", line 299, in set_input_data_dict
    df_weather = fcst.get_weather_forecast(
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/src/emhass/forecast.py", line 272, in get_weather_forecast
    data_15min.index = data_15min.index.tz_localize(self.time_zone)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/.venv/lib/python3.12/site-packages/pandas/core/indexes/datetimes.py", line 293, in tz_localize
    arr = self._data.tz_localize(tz, ambiguous, nonexistent)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/.venv/lib/python3.12/site-packages/pandas/core/arrays/_mixins.py", line 81, in method
    return meth(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/.venv/lib/python3.12/site-packages/pandas/core/arrays/datetimes.py", line 1088, in tz_localize
    new_dates = tzconversion.tz_localize_to_utc(
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "tzconversion.pyx", line 431, in pandas._libs.tslibs.tzconversion.tz_localize_to_utc
pytz.exceptions.NonExistentTimeError: 2025-03-30 02:00:00

**To Reproduce**
Choose a timezone with DST in my case Europe/Amsterdam and choose 'open-meteo' for weather forcecasts

**Expected behavior**
Ignore the missing hour when switching to DST

**Home Assistant installation type**
 - Home Assistant OS 
 -
**Your hardware**
- OS: HA OS,
- Architecture: aarch64
- O-Droid N2

**EMHASS installation type**
 - Add-on

**Additional context**
Thanks for the tool, really happy using it!

davidusb-geek, 2025-03-28:
Yes I had the same problem today. Change the method to something else just for today and tomorrow. The solar.forecast works fine. We know how to solve this and it will be fixed, just forgot to apply the correction for DST on the new method

han-alink, 2025-03-29:
Using solar forecast also results in an error:
File "/app/src/emhass/forecast.py", line 477, in get_weather_forecast
    "ts": list(data_raw["result"]["watts"].keys()),
               ~~~~~~~~~~~~~~~~~~^^^^^^^^^
TypeError: string indices must be integers, not 'str'

davidusb-geek, 2025-03-29:
The error is quite explicit, you need to use an integer value instead of a string.
Solar.forecast works perfectly fine on DST

han-alink, 2025-03-29:
As you suggested, I only changed from meteo forecast to solar forecast. I have defined my solar installation by setting the details of the pannels and inverters. With meteo forecast emhass is operating normal.

han-alink, 2025-03-30:
Still complaining about the dat/time (that is passed in meantime).

llevering, 2025-03-31:
> Still complaining about the dat/time (that is passed in meantime).

It loads and processes the weather information for the whole day. Everything is back to normal now.

purcell-lab, 2025-04-05:
Lots of folks in Australia getting hit with the DST bug today..

Hopefully this should clear in the next 24 hours, once the time jump passes.

closed 2025-08-27 by davidusb-geek