davidusb-geek/emhass #500: Error in v0.13.0 KeyError: 'sensor.power_photovoltaics'

philjohn, 2025-04-03:
**Describe the bug**

Just did a fresh install of v0.13.0 (was previously on 0.10.6). When running a dayahead, or MPC optim I get the following log output, saying there's a KeyError:

```
[2025-04-03 21:41:07 +0100] [10] [INFO]  >> Obtaining params: 
[2025-04-03 21:41:07 +0100] [10] [INFO] Passed runtime parameters: {}
[2025-04-03 21:41:07 +0100] [10] [INFO]  >> Setting input data dict
[2025-04-03 21:41:07 +0100] [10] [INFO] Setting up needed data
[2025-04-03 21:41:07 +0100] [10] [INFO] Retrieve hass get data method initiated...
[2025-04-03 21:41:08 +0100] [10] [INFO] Retrieving weather forecast data using method = open-meteo
[2025-04-03 21:41:09 +0100] [10] [INFO] Adjusting PV forecast, retrieving history data for model fit
[2025-04-03 21:41:09 +0100] [10] [INFO] Retrieve hass get data method initiated...
[2025-04-03 21:41:14 +0100] [10] [ERROR] Exception on /action/naive-mpc-optim [POST]
Traceback (most recent call last):
  File "/app/.venv/lib/python3.12/site-packages/pandas/core/indexes/base.py", line 3805, in get_loc
    return self._engine.get_loc(casted_key)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "index.pyx", line 167, in pandas._libs.index.IndexEngine.get_loc
  File "index.pyx", line 196, in pandas._libs.index.IndexEngine.get_loc
  File "pandas/_libs/hashtable_class_helper.pxi", line 7081, in pandas._libs.hashtable.PyObjectHashTable.get_item
  File "pandas/_libs/hashtable_class_helper.pxi", line 7089, in pandas._libs.hashtable.PyObjectHashTable.get_item
KeyError: 'sensor.power_photovoltaics'
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/app/.venv/lib/python3.12/site-packages/flask/app.py", line 1511, in wsgi_app
    response = self.full_dispatch_request()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/.venv/lib/python3.12/site-packages/flask/app.py", line 919, in full_dispatch_request
    rv = self.handle_user_exception(e)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/.venv/lib/python3.12/site-packages/flask/app.py", line 917, in full_dispatch_request
    rv = self.dispatch_request()
         ^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/.venv/lib/python3.12/site-packages/flask/app.py", line 902, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/src/emhass/web_server.py", line 414, in action_call
    input_data_dict = set_input_data_dict(
                      ^^^^^^^^^^^^^^^^^^^^
  File "/app/src/emhass/command_line.py", line 356, in set_input_data_dict
    P_PV_forecast = adjust_pv_forecast(
                    ^^^^^^^^^^^^^^^^^^^
  File "/app/src/emhass/command_line.py", line 129, in adjust_pv_forecast
    fcst.adjust_pv_forecast_data_prep(df_input_data)
  File "/app/src/emhass/forecast.py", line 756, in adjust_pv_forecast_data_prep
    P_PV = data["sensor.power_photovoltaics"]  # Actual PV production
           ~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/.venv/lib/python3.12/site-packages/pandas/core/frame.py", line 4102, in __getitem__
    indexer = self.columns.get_loc(key)
              ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/.venv/lib/python3.12/site-packages/pandas/core/indexes/base.py", line 3812, in get_loc
    raise KeyError(key) from err
KeyError: 'sensor.power_photovoltaics'
```

**To Reproduce**

Fresh install, simply defined power plant, and HA sensors for photovoltaic power. Secrets is correctly populated with HA token, lat, lon and elevation

**Expected behavior**

Should get a dayahead optimisation

**Home Assistant installation type**
 - Home Assistant Core

**Your hardware**
- OS: Official HA docker container
- Architecture: amd64

**EMHASS installation type**
 - Docker Standalone

**Additional context**

Config:

```
{
    "logging_level": "INFO",
    "costfun": "profit",
    "optimization_time_step": 30,
    "historic_days_to_retrieve": 9,
    "method_ts_round": "last",
    "continual_publish": false,
    "data_path": "default",
    "set_total_pv_sell": false,
    "lp_solver": "default",
    "lp_solver_path": "empty",
    "set_nocharge_from_grid": false,
    "set_nodischarge_to_grid": false,
    "set_battery_dynamic": false,
    "battery_dynamic_max": 0.9,
    "battery_dynamic_min": -0.9,
    "weight_battery_discharge": 1,
    "weight_battery_charge": 1,
    "sensor_power_photovoltaics": "sensor.solar_panel_production_w",
    "sensor_power_photovoltaics_forecast": "",
    "sensor_power_load_no_var_loads": "sensor.solar_house_consumption_w",
    "sensor_replace_zero": [
        "sensor.solar_panel_production_w",
        "sensor.p_pv_forecast"
    ],
    "sensor_linear_interp": [
        "sensor.solar_panel_production_w",
        "sensor.power_load_no_var_loads"
    ],
    "load_negative": false,
    "set_zero_min": true,
    "number_of_deferrable_loads": 1,
    "nominal_power_of_deferrable_loads": [
        3000
    ],
    "minimum_power_of_deferrable_loads": [
        0
    ],
    "operating_hours_of_each_deferrable_load": [
        4
    ],
    "weather_forecast_method": "open-meteo",
    "load_forecast_method": "mlforecaster",
    "delta_forecast_daily": 1,
    "load_cost_forecast_method": "hp_hc_periods",
    "start_timesteps_of_each_deferrable_load": [
        0
    ],
    "end_timesteps_of_each_deferrable_load": [
        0
    ],
    "load_peak_hour_periods": {
        "period_hp_1": [
            {
                "start": "05:30"
            },
            {
                "end": "23:30"
            }
        ]
    },
    "treat_deferrable_load_as_semi_cont": [
        true
    ],
    "set_deferrable_load_single_constant": [
        false
    ],
    "set_deferrable_startup_penalty": [
        0
    ],
    "load_peak_hours_cost": 0.2777,
    "load_offpeak_hours_cost": 0.07,
    "production_price_forecast_method": "constant",
    "photovoltaic_production_sell_price": 0.15,
    "maximum_power_from_grid": 22000,
    "maximum_power_to_grid": 5200,
    "set_use_pv": true,
    "set_use_adjusted_pv": true,
    "adjusted_pv_regression_model": "LassoRegression",
    "adjusted_pv_solar_elevation_threshold": 10,
    "pv_module_model": [
        "Maxeon_Solar_Technologies_Ltd__SPR_MAX6_405",
        "Maxeon_Solar_Technologies_Ltd__SPR_MAX6_405",
        "Maxeon_Solar_Technologies_Ltd__SPR_MAX6_405"
    ],
    "pv_inverter_model": [
        "SolarEdge_Technologies_Ltd___SE10000A_US__240V_",
        "SolarEdge_Technologies_Ltd___SE10000A_US__240V_",
        "SolarEdge_Technologies_Ltd___SE10000A_US__240V_"
    ],
    "surface_tilt": [
        42,
	42,
	42
    ],
    "surface_azimuth": [
        64.5,
	248,
	160
    ],
    "modules_per_string": [
        16,
	13,
	3
    ],
    "strings_per_inverter": [
        1,
	1,
	1
    ],
    "inverter_is_hybrid": true,
    "compute_curtailment": true,
    "set_use_battery": true,
    "battery_discharge_power_max": 5000,
    "battery_charge_power_max": 5000,
    "battery_discharge_efficiency": 0.95,
    "battery_charge_efficiency": 0.95,
    "battery_nominal_energy_capacity": 9700,
    "battery_minimum_state_of_charge": 0,
    "battery_maximum_state_of_charge": 1,
    "battery_target_state_of_charge": 0.6
}
```

philjohn, 2025-04-03:
OK, adding some logging to grab the content of the dataframe, I see that it contains the following keys:

```
[2025-04-03 21:56:52 +0100] [902] [INFO]                            sensor.solar_panel_production_w  sensor.solar_house_consumption_w_positive
2025-03-25 00:00:00+00:00                         5.164610                                2848.050293
2025-03-25 00:30:00+00:00                         4.774072                                2810.968190
2025-03-25 01:00:00+00:00                         2.919503                                2864.026621
2025-03-25 01:30:00+00:00                         3.610633                                2956.838435
2025-03-25 02:00:00+00:00                         4.256651                                2861.474095
...                                                    ...                                        ...
2025-04-03 19:30:00+01:00                         8.697297                                 746.863913
2025-04-03 20:00:00+01:00                         2.380000                                 695.406773
2025-04-03 20:30:00+01:00                        23.763636                                1017.729017
2025-04-03 21:00:00+01:00                        32.138710                                3417.432671
2025-04-03 21:30:00+01:00                         6.367547                                1246.465352
```

but the optimisation step is looking for `sensor.power_photovoltaics`

davidusb-geek, 2025-04-03:
Just set set_use_adjusted_pv=false
Sorry about this, its a current bug that is already fixed

philjohn, 2025-04-03:
Ah, yes, I just did that and was coming back to say that solved it, no worries, some great new features, will be great to try them out soon :)

closed 2025-04-03 by philjohn