davidusb-geek/emhass #504: 0.13 breaks mloptimiser

scruysberghs, 2025-04-05:
**Describe the bug**
I run mpc optimiser every couple of minutes and do a dairly fit to regenerate the ml prediction model. After upgrading to 0.13 running mpc with the mlforecaster gives me an error. Just checked, the fit ran succesfull.

 [2025-04-05 13:20:05 +0200] [23] [ERROR] The ML forecaster file was not found, please run a model fit method before this predict method

**To Reproduce**
Steps to reproduce the behavior

run the mpc opti;iser with the mlforecaster

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Home Assistant installation type**
 - Home Assistant OS
 - Home Assistant Supervised
 - Home Assistant Core

**Your hardware**
- OS: HA OS, Windows, Linux, etc
- Architecture: amd64, armhf, armv7, aarch64

**EMHASS installation type**
 - Add-on, Docker Standalone, Legacy Python virtual environment
add-on

**Additional context**
Add any other context about the problem here.

switching to navive for noz\w as workaround

hendrer, 2025-04-06:
Same exact issue here. Downgraded to 0.128

davidusb-geek, 2025-04-06:
Weird! What are your settings for the data folder path?
Fr now I don't how the new features could have broken this part?

hendrer, 2025-04-06:
You might be on to something: /share for the data folder path for me

davidusb-geek, 2025-04-06:
Try changing to the default.

hendrer, 2025-04-06:
I did - did a model fit and then predict, works like a charm but MPC continues to produce this: [2025-04-06 14:55:34 +0200] [27] [ERROR] The ML forecaster file was not found, please run a model fit method before this predict method
[2025-04-06 14:55:34 +0200] [27] [ERROR] Unable to get sensor power photovoltaics, or sensor power load no var loads. Check HA sensors and their daily data

json config:
{
  "adjusted_pv_regression_model": "LassoRegression",
  "adjusted_pv_solar_elevation_threshold": 10,
  "battery_charge_efficiency": 0.96,
  "battery_charge_power_max": 5200,
  "battery_discharge_efficiency": 0.96,
  "battery_discharge_power_max": 7200,
  "battery_dynamic_max": 0.9,
  "battery_dynamic_min": -0.9,
  "battery_maximum_state_of_charge": 1,
  "battery_minimum_state_of_charge": 0.06,
  "battery_nominal_energy_capacity": 19200,
  "battery_target_state_of_charge": 0.06,
  "compute_curtailment": false,
  "continual_publish": true,
  "costfun": "profit",
  "delta_forecast_daily": 1,
  "end_timesteps_of_each_deferrable_load": [
    0
  ],
  "historic_days_to_retrieve": 450,
  "inverter_is_hybrid": true,
  "load_cost_forecast_method": "hp_hc_periods",
  "load_forecast_method": "mlforecaster",
  "load_negative": false,
  "load_offpeak_hours_cost": 0.1419,
  "load_peak_hour_periods": {
    "period_hp_1": [
      {
        "start": "02:54"
      },
      {
        "end": "15:24"
      }
    ],
    "period_hp_2": [
      {
        "start": "17:24"
      },
      {
        "end": "20:24"
      }
    ]
  },
  "load_peak_hours_cost": 0.1907,
  "logging_level": "WARNING",
  "lp_solver": "default",
  "lp_solver_path": "empty",
  "maximum_power_from_grid": 11000,
  "maximum_power_to_grid": 11000,
  "method_ts_round": "first",
  "minimum_power_of_deferrable_loads": [
    0,
    0
  ],
  "modules_per_string": [
    16
  ],
  "nominal_power_of_deferrable_loads": [
    1
  ],
  "number_of_deferrable_loads": 1,
  "operating_hours_of_each_deferrable_load": [
    1
  ],
  "optimization_time_step": 60,
[... 128 lines cut ...]
2025-04-07 12:00:00+02:00    4917
2025-04-07 13:00:00+02:00    5145
dtype: int64
[2025-04-06 14:59:57 +0200] [27] [INFO] Retrieving data from hass for load forecast using method = mlforecaster
[2025-04-06 14:59:57 +0200] [27] [INFO] Retrieve hass get data method initiated...
[2025-04-06 15:00:00 +0200] [27] [INFO]  >> Obtaining params: 
[2025-04-06 15:00:00 +0200] [27] [INFO] Passed runtime parameters: {}
[2025-04-06 15:00:00 +0200] [27] [INFO]  >> Setting input data dict
[2025-04-06 15:00:00 +0200] [27] [INFO] Setting up needed data
[2025-04-06 15:00:00 +0200] [27] [INFO] Continual publish thread service started
[2025-04-06 15:00:00 +0200] [27] [INFO]  >> Publishing data...
[2025-04-06 15:00:00 +0200] [27] [INFO] Publishing data to HASS instance
[2025-04-06 15:00:00 +0200] [27] [INFO] Successfully posted to sensor.p_pv_forecast = 1412.0
[2025-04-06 15:00:00 +0200] [27] [INFO] Successfully posted to sensor.p_load_forecast = 1570.39
[2025-04-06 15:00:00 +0200] [27] [INFO] Successfully posted to sensor.p_hybrid_inverter = 1412.0
[2025-04-06 15:00:00 +0200] [27] [INFO] Successfully posted to sensor.p_deferrable0 = 0.0
[2025-04-06 15:00:00 +0200] [27] [INFO] Successfully posted to sensor.p_batt_forecast = 0.0
[2025-04-06 15:00:00 +0200] [27] [INFO] Successfully posted to sensor.soc_batt_forecast = 6.0
[2025-04-06 15:00:00 +0200] [27] [INFO] Successfully posted to sensor.p_grid_forecast = 158.39
[2025-04-06 15:00:00 +0200] [27] [INFO] Successfully posted to sensor.total_cost_fun_value = -10.25
[2025-04-06 15:00:00 +0200] [27] [INFO] Successfully posted to sensor.optim_status = Optimal
[2025-04-06 15:00:00 +0200] [27] [INFO] Successfully posted to sensor.unit_load_cost = 0.2786
[2025-04-06 15:00:00 +0200] [27] [INFO] Successfully posted to sensor.unit_prod_price = 0.2786
[2025-04-06 15:00:05 +0200] [27] [ERROR] The ML forecaster file was not found, please run a model fit method before this predict method
[2025-04-06 15:00:05 +0200] [27] [ERROR] Unable to get sensor power photovoltaics, or sensor power load no var loads. Check HA sensors and their daily data

martinarva, 2025-04-07:
Same here with 0.13 and mlforecaster. Model fit works, but when running MPC im get the "The ML forecaster file was not found, please run a model fit method before this predict method"

hendrer, 2025-04-08:
Just to confirm, reverting back to 12.8 works perfectly with the same config.

scruysberghs, 2025-04-08:
just had another look. EMHASS 0.13, addon set to store data in the "default" location.

[image]

Running a model fit creates the "load_forecast_mlf.pkl" file in the /data folder

[image]

Running a mpc optimiser results in:
```
  'load_forecast_method': 'mlforecaster'}
INFO - emhass.web_server -  >> Setting input data dict
INFO - emhass.web_server - Setting up needed data
INFO - emhass.web_server - Retrieve hass get data method initiated...
INFO - emhass.web_server - Retrieving weather forecast data using method = list
INFO - emhass.web_server - Retrieving data from hass for load forecast using method = mlforecaster
INFO - emhass.web_server - Retrieve hass get data method initiated...
ERROR - emhass.web_server - The ML forecaster file was not found, please run a model fit method before this predict method
```
Tried copying the "load_forecast_mlf.pkl" file from /data to /share and reran mpc but still same error.

scruysberghs, 2025-04-08:
tried again with the location set to /data and a new fit, same result.

davidusb-geek, 2025-04-12:
Hi, there is no error traceback for this? Just that logged error message?

scruysberghs, 2025-04-12:
> Hi, there is no error traceback for this? Just that logged error message?

that is correct, only the error message about the ML Forecaster file followed by [ERROR] Unable to get sensor power photovoltaics, or sensor power load no var loads. Check HA sensors and their daily data

Maybe related to not using the default sensor names for sensor_power_load_no_var_loads and sensor_power_photovoltaics in ha? 
(Just guessing. I'm on holliday so didn't have the time to look at the code changes from 0.12.8 to 0.13)

davidusb-geek, 2025-04-12:
For now I'm not able to reproduce this.
This is the code snippet that loads the saved model:
```
            model_type = self.params["passed_data"]["model_type"]
            filename = model_type + "_mlf.pkl"
            filename_path = self.emhass_conf["data_path"] / filename
            if not debug:
                if filename_path.is_file():
                    with open(filename_path, "rb") as inp:
                        mlf = pickle.load(inp)
                else:
                    self.logger.error(
                        "The ML forecaster file was not found, please run a model fit method before this predict method"
                    )
                    return False
```
So this will depend on the `model_type` parameter.
Could you please test this by passing a `mode_type` argument at runtime when fitting the model and then again when launching the MPC?
Maybe it is a problem with the default value name for this parameter

davidusb-geek, 2025-04-12:
Just created a new special unit test for this here PR #509
But the test is passing fine.
It launch MPC with mlforecaster method

purcell-lab, 2025-04-13:
I have been having this error.

I suspect the name for the load_forecast model has been updated to long_train_data_mlf.pkl

Running the MLForecaster from the GUI works and creates a long_train_data model, so I suspect the GUI is saving where the optimisation is looking. (share/long_train_data_mlf.pkl) 

```
[2025-04-13 09:49:47 +1000] [23] [INFO]  >> Performing a machine learning forecast model fit...
[2025-04-13 09:49:47 +1000] [23] [INFO] Performing a forecast model fit for long_train_data
[2025-04-13 09:49:47 +1000] [23] [INFO] Training a KNeighborsRegressor model
[2025-04-13 09:49:47 +1000] [23] [INFO] Elapsed time for model fit: 0.05205035209655762
[2025-04-13 09:49:47 +1000] [23] [INFO] Prediction R2 score of fitted model on test data: -0.17132995882779078
[2025-04-13 09:49:48 +1000] [23] [INFO]  >> Sending rendered template table data
...
[2025-04-13 09:50:29 +1000] [23] [INFO] Retrieving data from hass for load forecast using method = mlforecaster
[2025-04-13 09:50:29 +1000] [23] [INFO] Retrieve hass get data method initiated...
[2025-04-13 09:50:33 +1000] [23] [DEBUG] Number of ML predict forcast data generated (lags_opt): 48
[2025-04-13 09:50:33 +1000] [23] [DEBUG] Number of forcast dates obtained: 48
```

[image]

Updating my rest command to use the model long_train_data (from "model_type": "load_forecast") seems to be working:

```yaml
  forecast_model_fit:
    url: http://localhost:5000/action/forecast-model-fit
    method: POST
    timeout: 120
    content_type: "application/json"
    payload: >-
      {
        "var_model": "sensor.power_load_no_var_loads",
        "num_lags": 96,
        "split_date_delta": "48h",
        "model_type": "long_train_data",
        "optimization_time_step": 30,
        "sklearn_model": "KNeighborsRegressor",
        "fit_intercept": "True",
        "perform_backtest": "False"
      }
```

scruysberghs, 2025-04-13:
For reference. I pass these parameters to my model fit script:
````
      - action: script.emhass_ml_model_fit_script
        continue_on_error: true
        metadata: {}
        data:
          payload: >
            {
              "historic_days_to_retrieve": 30,
              "model_type": "load_forecast",
              "var_model": "sensor.sec_pac_non_deferable",
              "sklearn_model": "KNeighborsRegressor",
              "num_lags": 48,
              "split_date_delta": "48h",
              "perform_backtest": false
              }
```` 
This generates a file called load_forecast_mlf.pkl in the /data folder.
I didn't have a model_type in my mpc runtime parameters.
If I add 'model_type': 'load_forecast' to my mpc runtime parameters it runs fine.

This led me to another small issue with the new adjusted_pv functionality. Looks like sensor_power_photovoltaics  is hard coded to "sensor.power_photovoltaics" instead of using the runtime parameters or configured value.

````
INFO - emhass.web_server -  >> Obtaining params:
INFO - emhass.web_server - Passed runtime parameters: {'entity_save': True, 'publish_prefix': 'mpc_', 'prediction_horizon': 29, 'load_cost_forecast': [0.0986, 0.0894, 0.0894, 0.0894, 0.0894, 0.0873, 0.0873, 0.0691, 0.0691, 0.062, 0.062, 0.0787, 0.0787, 0.0894, 0.0894, 0.1072, 0.1072, 0.1801, 0.1801, 0.1989, 0.1989, 0.2092, 0.2092, 0.2063, 0.2063, 0.1906, 0.1906, 0.1837, 0.1837], 'prod_price_forecast': [-0.0067, -0.015, -0.015, -0.015, -0.015, -0.017, -0.017, -0.0335, -0.0335, -0.0399, -0.0399, -0.0247, -0.0247, -0.015, -0.015, 0.0011, 0.0011, 0.0672, 0.0672, 0.0842, 0.0842, 0.0936, 0.0936, 0.0909, 0.0909, 0.0767, 0.0767, 0.0704, 0.0704], 'pv_power_forecast': [431, 194, 211, 226, 241, 276, 328, 273, 124, 82, 129, 188, 248, 322, 384, 331, 218, 127, 62, 21, 2, 0, 0, 0, 0, 0, 0, 0, 0], 'soc_init': 0.12, 'soc_final': 0.6, 'alpha': 0, 'beta': 1, 'optimization_time_step': 30, 'historic_days_to_retrieve': 9, 'sensor_power_photovoltaics': 'sensor.sec_pac_solar', 'sensor_power_load_no_var_loads': 'sensor.sec_pac_non_deferable', 'sensor_linear_interp': ['sensor.sec_pac_non_deferable', 'sensor.sec_pac_solar'], 'num_def_loads': 4, 'operating_hours_of_each_deferrable_load': [0, 0, 0, 0], 'def_current_state': [False, False, False, False], 'nominal_power_of_deferrable_loads': [0.0, 0.0, 0.0, 0.0], 'start_timesteps_of_each_deferrable_load': [0, 0, 0, 0], 'end_timesteps_of_each_deferrable_load': [0, 0, 0, 0], 'treat_deferrable_load_as_semi_cont': [True, False, True, True], 'set_deferrable_load_single_constant': [True, False, True, True], 'set_deferrable_startup_penalty': [0, 0, 0, 0], 'maximum_power_from_grid': 2500, 'maximum_power_to_grid': 7000, 'inverter_is_hybrid': False, 'compute_curtailment': True, 'battery_discharge_power_max': 2500, 'battery_charge_power_max': 2500, 'battery_discharge_efficiency': 0.96, 'battery_charge_efficiency': 0.96, 'battery_nominal_energy_capacity': 5120, 'battery_maximum_state_of_charge': 0.98, 'battery_minimum_state_of_charge': 0.16, 'battery_target_state_of_charge': 0.6, 'weight_battery_discharge': 0.01, 'weight_battery_charge': 0.01, 'set_battery_dynamic': False, 'battery_dynamic_max': 0.9, 'battery_dynamic_min': -0.9, 'set_nocharge_from_grid': False, 'set_nodischarge_to_grid': False, 'costfun': 'profit', 'lp_solver': 'default', 'load_forecast_method': 'mlforecaster', 'model_type': 'load_forecast'}
INFO - emhass.web_server -  >> Setting input data dict
INFO - emhass.web_server - Setting up needed data
WARNING - emhass.web_server - Passed soc_init=0.12 is lower than soc_min=0.16, setting soc_init=soc_min
INFO - emhass.web_server - Retrieve hass get data method initiated...
INFO - emhass.web_server - Retrieving weather forecast data using method = list
DEBUG - emhass.web_server - get_weather_forecast returning:
                           yhat
ts
2025-04-13 10:00:00+02:00   431
2025-04-13 10:30:00+02:00   194
2025-04-13 11:00:00+02:00   211
2025-04-13 11:30:00+02:00   226
2025-04-13 12:00:00+02:00   241
2025-04-13 12:30:00+02:00   276
2025-04-13 13:00:00+02:00   328
2025-04-13 13:30:00+02:00   273
2025-04-13 14:00:00+02:00   124
2025-04-13 14:30:00+02:00    82
2025-04-13 15:00:00+02:00   129
2025-04-13 15:30:00+02:00   188
2025-04-13 16:00:00+02:00   248
2025-04-13 16:30:00+02:00   322
2025-04-13 17:00:00+02:00   384
2025-04-13 17:30:00+02:00   331
2025-04-13 18:00:00+02:00   218
2025-04-13 18:30:00+02:00   127
2025-04-13 19:00:00+02:00    62
2025-04-13 19:30:00+02:00    21
2025-04-13 20:00:00+02:00     2
2025-04-13 20:30:00+02:00     0
2025-04-13 21:00:00+02:00     0
2025-04-13 21:30:00+02:00     0
2025-04-13 22:00:00+02:00     0
2025-04-13 22:30:00+02:00     0
2025-04-13 23:00:00+02:00     0
2025-04-13 23:30:00+02:00     0
2025-04-14 00:00:00+02:00     0
DEBUG - emhass.web_server - get_power_from_weather returning:
ts
2025-04-13 10:00:00+02:00    378
2025-04-13 10:30:00+02:00    194
2025-04-13 11:00:00+02:00    211
2025-04-13 11:30:00+02:00    226
2025-04-13 12:00:00+02:00    241
[... 51 lines cut ...]
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/src/emhass/web_server.py", line 414, in action_call
    input_data_dict = set_input_data_dict(
                      ^^^^^^^^^^^^^^^^^^^^
  File "/app/src/emhass/command_line.py", line 356, in set_input_data_dict
    P_PV_forecast = adjust_pv_forecast(
[... cut ...]
    indexer = self.columns.get_loc(key)
              ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/.venv/lib/python3.12/site-packages/pandas/core/indexes/base.py", line 3812, in get_loc
    raise KeyError(key) from err
KeyError: 'sensor.power_photovoltaics'

````

@davidusb-geek let me know if you want me to make a seperate issue for this. Searched the repo but could not find the hard coded value anywhere but in default values and tests.

My workaround is to add` 'set_use_adjusted_pv': False` to the mpc parameters for now.

davidusb-geek, 2025-04-13:
Ok great. The name of your model needs to be specified in both the fit and mpc methods.
Yes for the second issue it has been identified and already fixed here #499

scruysberghs, 2025-04-13:
> I have been having this error.
> 
> I suspect the name for the load_forecast model has been updated to long_train_data_mlf.pkl
> 
> Running the MLForecaster from the GUI works and creates a long_train_data model, so I suspect the GUI is saving where the optimisation is looking. (share/long_train_data_mlf.pkl)
> 
> ```
> [2025-04-13 09:49:47 +1000] [23] [INFO]  >> Performing a machine learning forecast model fit...
> [2025-04-13 09:49:47 +1000] [23] [INFO] Performing a forecast model fit for long_train_data
> [2025-04-13 09:49:47 +1000] [23] [INFO] Training a KNeighborsRegressor model
> [2025-04-13 09:49:47 +1000] [23] [INFO] Elapsed time for model fit: 0.05205035209655762
> [2025-04-13 09:49:47 +1000] [23] [INFO] Prediction R2 score of fitted model on test data: -0.17132995882779078
> [2025-04-13 09:49:48 +1000] [23] [INFO]  >> Sending rendered template table data
> ...
> [2025-04-13 09:50:29 +1000] [23] [INFO] Retrieving data from hass for load forecast using method = mlforecaster
> [2025-04-13 09:50:29 +1000] [23] [INFO] Retrieve hass get data method initiated...
> [2025-04-13 09:50:33 +1000] [23] [DEBUG] Number of ML predict forcast data generated (lags_opt): 48
> [2025-04-13 09:50:33 +1000] [23] [DEBUG] Number of forcast dates obtained: 48
> ```
> 
> [image]
> 
> Updating my rest command to use the model long_train_data (from "model_type": "load_forecast") seems to be working:
> 
>   forecast_model_fit:
>     url: http://localhost:5000/action/forecast-model-fit
>     method: POST
>     timeout: 120
>     content_type: "application/json"
>     payload: >-
>       {
>         "var_model": "sensor.power_load_no_var_loads",
>         "num_lags": 96,
>         "split_date_delta": "48h",
>         "model_type": "long_train_data",
>         "optimization_time_step": 30,
>         "sklearn_model": "KNeighborsRegressor",
>         "fit_intercept": "True",
>         "perform_backtest": "False"
>       }

Hi Mark @purcell-lab , 
Unrelated question. I see in your screenshot that you use emhass ML forecaster to generate cost and price predictions. Is that dynamic electricity  pricing prediction only looking at recent history or is there a way to look at covariants like predicted solar/wind/weather. How accurate can you get this where you live? 
I am currently looking at some options to do some multiple day optimisations in emhass but am still looking at the best way to have more than the max 13 to 36hrs of actual published pricing here in EU-West . Also played around with the ML fit in emhass but the results weren't usable here (just using pricing from the day before was more accurate).

I am now doing some experiments with the LEAR model https://epftoolbox.readthedocs.io/en/latest/modules/models.html

But maybe we should start another thread on this....

purcell-lab, 2025-04-13:
> ML forecaster to generate cost and price predictions

Yes I'm definitely having some fun generating multi day f forecasts. 

[image]

davidusb-geek, 2025-04-21:
I'll close this issue. The solution is to specify `var_model` on your fit and predict calls.
If you think its needed feel free to re-open

closed 2025-04-21 by davidusb-geek