davidusb-geek/emhass #516: Problem with Naive Load forecast

radim2tej, 2025-04-22:
**Describe the bug**
For the new versions 0.13.0 and 0.13.1, the Naive Load forecast is shifted. It looks like the system does not take the load start as the current time in UTC - 24h, but the time 0:00 of the previous day. In version 0.12.7 it worked ok. In the picture the home load is the blue line.

**Screenshots**
0.12.7
[image]

0.13.x
[image]

System
- Core 2025.4.2
- Supervisor 2025.04.0
- Operating System 15.2
- OS: HA OS
- Architecture: amd64
- Add-on

purcell-lab, 2025-04-23:
I agree with this issue.

Naive forecast (blue load line) is +1000 out os sync with my actual historic load (black load ML forecast)

[image]

davidusb-geek, 2025-04-24:
Trying to solve this.
@purcell-lab Would you be able to test if this is solved using the test image from PR #522 ?

purcell-lab, 2025-04-25:
I have tried but I get the following error:

I'm sure the fix is simple, but I'm out of my depth.

```logs
[2025-04-25 17:30:32 +1000] [22] [ERROR] Exception on /action/naive-mpc-optim [POST]
Traceback (most recent call last):
  File "/app/.venv/lib/python3.12/site-packages/flask/app.py", line 1511, in wsgi_app
    response = self.full_dispatch_request()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/.venv/lib/python3.12/site-packages/flask/app.py", line 919, in full_dispatch_request
    rv = self.handle_user_exception(e)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/.venv/lib/python3.12/site-packages/flask/app.py", line 917, in full_dispatch_request
    rv = self.dispatch_request()
         ^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/.venv/lib/python3.12/site-packages/flask/app.py", line 902, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/src/emhass/web_server.py", line 469, in action_call
    opt_res = naive_mpc_optim(input_data_dict, app.logger)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/src/emhass/command_line.py", line 744, in naive_mpc_optim
    df_input_data_dayahead = input_data_dict["fcst"].get_load_cost_forecast(
                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/src/emhass/forecast.py", line 1592, in get_load_cost_forecast
    forecast_out = self.get_forecast_out_from_csv_or_list(
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/src/emhass/forecast.py", line 1159, in get_forecast_out_from_csv_or_list
    df_csv.loc[fcst_index, :]
    ~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/app/.venv/lib/python3.12/site-packages/pandas/core/indexing.py", line 1184, in __getitem__
    return self._getitem_tuple(key)
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/.venv/lib/python3.12/site-packages/pandas/core/indexing.py", line 1377, in _getitem_tuple
    return self._getitem_tuple_same_dim(tup)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/.venv/lib/python3.12/site-packages/pandas/core/indexing.py", line 1020, in _getitem_tuple_same_dim
    retval = getattr(retval, self.name)._getitem_axis(key, axis=i)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/.venv/lib/python3.12/site-packages/pandas/core/indexing.py", line 1420, in _getitem_axis
    return self._getitem_iterable(key, axis=axis)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/.venv/lib/python3.12/site-packages/pandas/core/indexing.py", line 1360, in _getitem_iterable
    keyarr, indexer = self._get_listlike_indexer(key, axis)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/.venv/lib/python3.12/site-packages/pandas/core/indexing.py", line 1558, in _get_listlike_indexer
    keyarr, indexer = ax._get_indexer_strict(key, axis_name)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/.venv/lib/python3.12/site-packages/pandas/core/indexes/base.py", line 6200, in _get_indexer_strict
    self._raise_if_missing(keyarr, indexer, axis_name)
  File "/app/.venv/lib/python3.12/site-packages/pandas/core/indexes/base.py", line 6249, in _raise_if_missing
    raise KeyError(f"None of [{key}] are in the [{axis_name}]")
KeyError: "None of [DatetimeIndex(['2025-04-25 00:00:00+10:00', '2025-04-25 00:30:00+10:00',\n               '2025-04-25 01:00:00+10:00', '2025-04-25 01:30:00+10:00',\n               '2025-04-25 02:00:00+10:00', '2025-04-25 02:30:00+10:00',\n               '2025-04-25 03:00:00+10:00', '2025-04-25 03:30:00+10:00',\n               '2025-04-25 04:00:00+10:00', '2025-04-25 04:30:00+10:00',\n               '2025-04-25 05:00:00+10:00', '2025-04-25 05:30:00+10:00',\n               '2025-04-25 06:00:00+10:00', '2025-04-25 06:30:00+10:00',\n               '2025-04-25 07:00:00+10:00', '2025-04-25 07:30:00+10:00',\n               '2025-04-25 08:00:00+10:00', '2025-04-25 08:30:00+10:00',\n               '2025-04-25 09:00:00+10:00', '2025-04-25 09:30:00+10:00',\n               '2025-04-25 10:00:00+10:00'],\n              dtype='datetime64[ns, Australia/Brisbane]', freq='30min')] are in the [index]"
```

davidusb-geek, 2025-04-29:
Ok sorry, I thought this was a very simple fix but I was wrong.
Will need some more time to debug

ceedeebee907, 2025-06-15:
Following, having the same issue here

AtomicGrog, 2025-06-26:
> Following, having the same issue here

Likewise. Would like to use some of the new solar prediction options but this is breaking my load history;

davidusb-geek, 2025-06-27:
> > Following, having the same issue here
> 
> Likewise. Would like to use some of the new solar prediction options but this is breaking my load history;

How is that?

AtomicGrog, 2025-06-27:
> > > Following, having the same issue here
> > 
> > 
> > Likewise. Would like to use some of the new solar prediction options but this is breaking my load history;
> 
> How is that?

Maybe i termed it incorrectly, when emhass runs the load history still shows out of sync i.e. here's two images one from 0.12.7 the other from 0.13.3 the 0.12.7 shows load during the evening period i.e. 5pm to 9.30pm the which is correct the 0.13.3 shows the same load profile but it's been brought forward in the day, or back, its really hard to comprehend but whatever is happening there should be load depicted between the evening period and way less in the afternoon.

[image]
[image]

tbrasser, 2025-07-24:
It seems the load forecast assumes that it runs between +-23:00 - 01:00

if I run it mid-day, the load forecast is shifted by that same amount of hours.

davidusb-geek, 2025-08-26:
> Any updates here? Are we supposed/expected to run only at midnight?

I'm on to debugging this but it doesn't seem to be that simple.
No, this is supposed to work at any time. It worked fine for a long time and since the beginning of the project. 
But somehow recently it broke, searching for the bug.

davidusb-geek, 2025-08-26:
I've just merged #522 that seems to fix the problem, at least on my testing.
Please test it on you side with the new upcoming release.

AtomicGrog, 2025-08-30:
Can confirm its working fine in 0.13.4.