davidusb-geek/emhass #519: 0.13.1 breaks my mpc optimiser : [ERROR] The retrieved JSON is empty, A sensor:sensor.p_pv_forecast may have 0 days of history

scruysberghs, 2025-04-23:
**Describe the bug**
A clear and concise description of what the bug is.
After the changes in 0.13 I had this in my mpc runtime parameters:
'load_forecast_method': 'mlforecaster', 'model_type': 'load_forecast', 'set_use_adjusted_pv': False

That worken fine but after upgrading to 0.13.1 I now get :

[ERROR] The retrieved JSON is empty, A sensor:sensor.p_pv_forecast may have 0 days of history, passed sensor may not be correct, or days to retrieve is set too heigh

Probably because I only publish with the mpc_ prefix. So my ha has a sensor called sensor.mpc_p_pv_forecast but no sensor called sensor.p_pv_forecast

paulhomes, 2025-04-23:
This sounds similar to https://github.com/davidusb-geek/emhass/issues/517

Is it resolved if you add the following to your config.json?

```
    "sensor_power_photovoltaics_forecast": "sensor.mpc_p_pv_forecast",
```

paulhomes, 2025-04-23:
@davidusb-geek I think this may be due to the following lines of code in command_line.py:

```
        if optim_conf.get("set_use_pv", True):
            var_list.append(retrieve_hass_conf["sensor_power_photovoltaics"])
            var_list.append(retrieve_hass_conf["sensor_power_photovoltaics_forecast"])
```

For those that are not using the new PV adjustment feature in 0.13 it adds the default config value for `sensor_power_photovoltaics_forecast` (`sensor.p_pv_forecast`) to the list of sensors to fetch from HA and fails if that sensor doesn't exist in HA.  Do you think it would be a good idea to change this to the following?

```
        if optim_conf.get("set_use_pv", True):
            var_list.append(retrieve_hass_conf["sensor_power_photovoltaics"])
        if optim_conf.get("set_use_adjusted_pv", True):
            var_list.append(retrieve_hass_conf["sensor_power_photovoltaics_forecast"])
```

scruysberghs, 2025-04-23:
adding "sensor_power_photovoltaics_forecast": "sensor.mpc_p_pv_forecast", to the runtime paramters does fix it.

Changing the default in the config would also have fixed it. 

But as you noticed since I had "set_use_adjusted_pv" set to false it wasn't supposed to be requested from HA. 

--
As 0.13.1 includes the fix to use a non default name for sensor_power_photovoltaics I'll switch to "set_use_adjusted_pv": true to try out this feature.
@davidusb-geek : a question related to predicted pv adjustments. Do you take curtailment into account? If curtailment is active the value of sensor_power_photovoltaics will be lower than predicted but those difference should not be used for the ML adjustments.

paulhomes, 2025-04-23:
Thanks for testing that. I've submitted a PR to ignore that config option when set_use_adjusted_pv is false.

davidusb-geek, 2025-05-02:
> [@davidusb-geek](https://github.com/davidusb-geek) : a question related to predicted pv adjustments. Do you take curtailment into account? If curtailment is active the value of sensor_power_photovoltaics will be lower than predicted but those difference should not be used for the ML adjustments.

For now the adjustment is based on historic values of both the real PV power and the forecasts. So it will consider any curtailment events as these are not filtered. But wouldn't it be nice to have a prediction of curtailment events integrated into the PV forecast?
We can open a discussion thread for this.

Closing this issue as solved

closed 2025-05-02 by davidusb-geek