davidusb-geek/emhass #521: Adjusted PV forecast can return negative PV values

paulhomes, 2025-04-24:
I have been running with set_use_adjusted_pv=true with 0.13.1 and today noticed that it returned a few negative values in p_pv_forecast, early in the morning and late in the afternoon.  Today has been a very cloudy/rainy day with a low PV forecast (<3kW) when recent history has been clear and sunny with much higher PV values (>6kW). 

I also have the following in config.json:

```
    "adjusted_pv_regression_model": "LassoRegression",
    "adjusted_pv_solar_elevation_threshold": 10,
```

Here is a debug log from a mpc run this morning showing the negative PV values:

```
emhass  | [2025-04-24 05:00:03 +1000] [24] [INFO]  >> Obtaining params:
emhass  | [2025-04-24 05:00:03 +1000] [24] [INFO] Passed runtime parameters: {'load_cost_forecast': [0.2, 0.21, 0.25, 0.24, 0.21, 0.18, 0.16, 0.16, 0.13, 0.13, 0.13, 0.13, 0.13, 0.13, 0.13, 0.13, 0.13, 0.15, 0.16, 0.13, 0.14, 0.17, 0.32, 0.38, 0.42, 0.65, 0.65, 0.6, 0.52, 0.44, 0.43, 0.41, 0.27, 0.26, 0.32, 0.31, 0.31, 0.29, 0.26, 0.25, 0.24, 0.24, 0.24, 0.23, 0.23, 0.22], 'prod_price_forecast': [0.08, 0.09, 0.12, 0.11, 0.09, 0.06, 0.04, 0.04, 0.04, 0.04, 0.04, 0.04, 0.04, 0.04, 0.04, 0.04, 0.04, 0.06, 0.07, 0.04, 0.04, 0.07, 0.07, 0.12, 0.16, 0.37, 0.37, 0.33, 0.26, 0.18, 0.17, 0.16, 0.15, 0.13, 0.19, 0.18, 0.18, 0.16, 0.13, 0.13, 0.12, 0.12, 0.11, 0.11, 0.11, 0.09], 'prediction_horizon': 46, 'soc_init': 0.5, 'weight_battery_discharge': 0.1, 'weight_battery_charge': 0.0}
emhass  | [2025-04-24 05:00:03 +1000] [24] [INFO]  >> Setting input data dict
emhass  | [2025-04-24 05:00:03 +1000] [24] [INFO] Setting up needed data
emhass  | [2025-04-24 05:00:03 +1000] [24] [INFO] Retrieve hass get data method initiated...
emhass  | [2025-04-24 05:00:04 +1000] [24] [DEBUG] prepare_data self.var_list=['sensor.load_power_less_deferrable', 'sensor.total_dc_power', 'sensor.p_pv_forecast']
emhass  | [2025-04-24 05:00:04 +1000] [24] [DEBUG] prepare_data var_load=sensor.load_power_less_deferrable
emhass  | [2025-04-24 05:00:04 +1000] [24] [DEBUG] prepare_data load_negative=False
emhass  | [2025-04-24 05:00:04 +1000] [24] [DEBUG] prepare_data set_zero_min=True
emhass  | [2025-04-24 05:00:04 +1000] [24] [DEBUG] prepare_data var_replace_zero=['sensor.total_dc_power', 'sensor.p_pv_forecast']
emhass  | [2025-04-24 05:00:04 +1000] [24] [DEBUG] prepare_data var_interp=['sensor.total_dc_power', 'sensor.load_power_less_deferrable']
emhass  | [2025-04-24 05:00:04 +1000] [24] [INFO] Retrieving weather forecast data using method = open-meteo
emhass  | [2025-04-24 05:00:04 +1000] [24] [INFO] Loading existing cached Open-Meteo JSON file: /data/cached-open-meteo-forecast.json
emhass  | [2025-04-24 05:00:04 +1000] [24] [INFO] The cached Open-Meteo JSON file is recent (age=24m, max_age=60m)
emhass  | [2025-04-24 05:00:04 +1000] [24] [DEBUG] get_weather_forecast returning:
emhass  |                            temp_air  relative_humidity  ...    dhi    dni
emhass  | 2025-04-24 05:00:00+10:00      18.0                 92  ...    0.0    0.0
emhass  | 2025-04-24 05:30:00+10:00      18.1                 93  ...    0.0    0.0
emhass  | 2025-04-24 06:00:00+10:00      18.1                 93  ...    0.0    0.0
emhass  | 2025-04-24 06:30:00+10:00      18.3                 93  ...   10.9   13.7
emhass  | 2025-04-24 07:00:00+10:00      18.5                 93  ...   29.7   57.7
emhass  | 2025-04-24 07:30:00+10:00      18.8                 93  ...   45.2   70.5
emhass  | 2025-04-24 08:00:00+10:00      19.2                 92  ...   65.2   79.3
emhass  | 2025-04-24 08:30:00+10:00      19.7                 90  ...   91.3   90.1
emhass  | 2025-04-24 09:00:00+10:00      20.2                 88  ...  118.2  120.4
emhass  | 2025-04-24 09:30:00+10:00      20.8                 86  ...  144.9  156.5
emhass  | 2025-04-24 10:00:00+10:00      21.3                 83  ...  163.2  167.5
emhass  | 2025-04-24 10:30:00+10:00      21.7                 80  ...  172.4  161.2
emhass  | 2025-04-24 11:00:00+10:00      22.0                 78  ...  176.5  138.0
emhass  | 2025-04-24 11:30:00+10:00      22.0                 78  ...  176.6  110.0
emhass  | 2025-04-24 12:00:00+10:00      22.0                 78  ...  194.7  123.3
emhass  | 2025-04-24 12:30:00+10:00      22.1                 78  ...  213.6  150.9
emhass  | 2025-04-24 13:00:00+10:00      22.1                 78  ...  193.7  117.2
emhass  | 2025-04-24 13:30:00+10:00      21.8                 79  ...  154.3   53.0
emhass  | 2025-04-24 14:00:00+10:00      21.4                 81  ...  130.8   21.6
emhass  | 2025-04-24 14:30:00+10:00      21.2                 82  ...  113.3    6.8
emhass  | 2025-04-24 15:00:00+10:00      21.2                 82  ...   99.6    3.9
emhass  | 2025-04-24 15:30:00+10:00      21.0                 83  ...   83.9    9.3
emhass  | 2025-04-24 16:00:00+10:00      20.8                 84  ...   59.7   11.9
emhass  | 2025-04-24 16:30:00+10:00      20.5                 86  ...   34.0   17.0
emhass  | 2025-04-24 17:00:00+10:00      20.2                 87  ...   10.0    7.1
emhass  | 2025-04-24 17:30:00+10:00      20.0                 88  ...    0.0    0.0
emhass  | 2025-04-24 18:00:00+10:00      19.9                 88  ...    0.0    0.0
emhass  | 2025-04-24 18:30:00+10:00      19.8                 89  ...    0.0    0.0
emhass  | 2025-04-24 19:00:00+10:00      19.6                 89  ...    0.0    0.0
emhass  | 2025-04-24 19:30:00+10:00      19.6                 89  ...    0.0    0.0
emhass  | 2025-04-24 20:00:00+10:00      19.5                 89  ...    0.0    0.0
emhass  | 2025-04-24 20:30:00+10:00      19.5                 89  ...    0.0    0.0
emhass  | 2025-04-24 21:00:00+10:00      19.4                 89  ...    0.0    0.0
emhass  | 2025-04-24 21:30:00+10:00      19.3                 88  ...    0.0    0.0
emhass  | 2025-04-24 22:00:00+10:00      19.2                 87  ...    0.0    0.0
emhass  | 2025-04-24 22:30:00+10:00      19.1                 87  ...    0.0    0.0
emhass  | 2025-04-24 23:00:00+10:00      19.0                 87  ...    0.0    0.0
emhass  | 2025-04-24 23:30:00+10:00      19.0                 87  ...    0.0    0.0
emhass  | 2025-04-25 00:00:00+10:00      18.8                 88  ...    0.0    0.0
emhass  | 2025-04-25 00:30:00+10:00      18.7                 89  ...    0.0    0.0
emhass  | 2025-04-25 01:00:00+10:00      18.6                 89  ...    0.0    0.0
emhass  | 2025-04-25 01:30:00+10:00      18.5                 89  ...    0.0    0.0
[... 155 lines cut ...]
emhass  | [2025-04-24 05:00:11 +1000] [24] [INFO] Total value of the Cost function = 0.13
emhass  | [2025-04-24 05:00:11 +1000] [24] [INFO]  >> Obtaining params:
emhass  | [2025-04-24 05:00:11 +1000] [24] [INFO] Passed runtime parameters: {}
emhass  | [2025-04-24 05:00:11 +1000] [24] [INFO]  >> Setting input data dict
emhass  | [2025-04-24 05:00:11 +1000] [24] [INFO] Setting up needed data
emhass  | [2025-04-24 05:00:12 +1000] [24] [INFO]  >> Publishing data...
emhass  | [2025-04-24 05:00:12 +1000] [24] [INFO] Publishing data to HASS instance
emhass  | [2025-04-24 05:00:12 +1000] [24] [INFO] Successfully posted to sensor.p_pv_forecast = 0.0
emhass  | [2025-04-24 05:00:12 +1000] [24] [INFO] Successfully posted to sensor.p_load_forecast = 412.0
emhass  | [2025-04-24 05:00:12 +1000] [24] [INFO] Successfully posted to sensor.p_pv_curtailment = 0.0
emhass  | [2025-04-24 05:00:12 +1000] [24] [INFO] Successfully posted to sensor.p_hybrid_inverter = 0.0
emhass  | [2025-04-24 05:00:12 +1000] [24] [INFO] Successfully posted to sensor.p_batt_forecast = 0.0
emhass  | [2025-04-24 05:00:12 +1000] [24] [INFO] Successfully posted to sensor.soc_batt_forecast = 50.0
emhass  | [2025-04-24 05:00:12 +1000] [24] [INFO] Successfully posted to sensor.p_grid_forecast = 412.0
emhass  | [2025-04-24 05:00:12 +1000] [24] [INFO] Successfully posted to sensor.total_cost_fun_value = 2.22
emhass  | [2025-04-24 05:00:12 +1000] [24] [INFO] Successfully posted to sensor.optim_status = Optimal
emhass  | [2025-04-24 05:00:12 +1000] [24] [INFO] Successfully posted to sensor.unit_load_cost = 0.2
emhass  | [2025-04-24 05:00:12 +1000] [24] [INFO] Successfully posted to sensor.unit_prod_price = 0.08
```

From debug logging, the negative values appear during the forecast.py `adjust_pv_forecast_predict` method in forecast_data.  There are no negative values in the results from get_power_from_weather.

I was wondering if the the apply_weighting function around line 969 might let some negatives through: https://github.com/davidusb-geek/emhass/blob/55be2aab4509dc8d5e2d738744cce3a6b0392348/src/emhass/forecast.py#L969

I was also wondering if I needed to adjust my adjusted_pv_solar_elevation_threshold config setting but am unsure how its optimal value is determined.

davidusb-geek, 2025-05-02:
Will try to find some time to look into this

paulhomes, 2025-05-04:
Thanks David. I was going to take a look myself but have not been able to replicate the negative values again yet. I probably need to wait for another cloudy/rainy day and capture the data for a unit test.

paulhomes, 2025-11-17:
I'm not sure if it was specifically solved, but I haven't seen it happen again for a very long time now so closing the issue is a good move. I'm also no longer using  set_use_adjusted_pv=true but an alternative external method for adjusting PV forecasts which I'll have a go at a PR for adding into emhass when I get some time.