davidusb-geek/emhass #533: Day_ahead optimization

renaatdb, 2025-05-01:
Hello,
When I use shell command with automation.

[image]

```
shell_command:
  dayahead_optim: "curl -i -H \"Content-Type:application/json\" -X POST -d '{\"pv_power_forecast\":{{states('sensor.solcast_24hrs_forecast')}}}' http://localhost:5000/action/dayahead-optim"
```
I get 

[image]

but when I push the Day-ahead Optmization button I see other results.
??

[image]

I do not understand.

davidusb-geek, 2025-05-01:
On your shell command you pass some custom data for the pv_power_forecast. But when you push the button on the webui that is using the methods that are fixed on your emhass configuration. So it seems that your PV forecast is different on two cases, hence the different results

renaatdb, 2025-05-28:
Ok thanks