davidusb-geek/emhass #552: [Feature Request] Flag in Deferrable Load Config to avoid Battery Usage when running

projectcobalt, 2025-05-27:
In an installation running in self-consumption mode there are certain times of the year where Max PV Output cannot completely cover the scheduled load, the optimal solution the controller finds is to provide the power gap from the battery.. 

This can be undesirable; B) when grid cost and battery soc is low B) The scheduled load is a HWS (why trade chemical energy for thermal energy.

I'm looking for a solution where I can specify where the power shortfall is supplied from, that led me to a possible solution being a flag that will constrain a scheduled load to PV/Grid.

However, I am open to any input on the best way to handle this situation outside of a new feature.

Octofinger, 2025-08-22:
Why don't you run in profit mode? That would probably give you the behaviour you want.

projectcobalt, 2025-08-27:
> Why don't you run in profit mode? That would probably give you the behaviour you want.

Have trialed all modes. Profit is not my primary goal - Generally based on my config emhass prefers to dump the battery at peak times and just run on grid until the solar window restarts...

My goals are;

Self-Consume
Lowest cost
Profit

So, in other words, draw from battery when no PV is available, run scheduled loads at optimal PV times and if PV does not cover full power requirement and grid is very cheap import the delta to achieve the lowest cost. If I have a sufficient battery capacity to self-consume and it is a peak time export just enough to generate some profit while maintaining the primary goal of self-consumption.

Noting that self-consumption does all that except for supplementing the PV | scheduled load delta with grid power especially when grid coast is less than discharge weight.

projectcobalt, 2025-11-13:
Hey @davidusb-geek here is another example which is quite interesting...

Looking from the "now" marker, Battery is discharged at start of load schedule then grid is called to recover it.. Start of load schedule grid cost is 7-8 cents then when grid is used it is 5 cents.

 I'm assuming the grid is used primarily because soc_final constraint would be enforcing recovery..

[image]

[opt_res_latest.csv](https://github.com/user-attachments/files/23535978/opt_res_latest.csv)

projectcobalt, 2025-11-16:
> do you pass max power from and to grid runtime parameters? battery charge and discharge weights? It feels like with those parameters you can avoid what you see hapening

Yeah I am, see below; Noting I have omitted load_power_forecast, pv_power_forecast, prod_price_forecast and load_cost_forecast from the below to keep condensed.

`[2025-11-16 17:25:19 +1000] [25] [INFO] Passed runtime parameters: {'costfun': 'self-consumption', 'prediction_horizon': 288, 'optimization_time_step': 5, 'number_of_deferrable_loads': 1, 'alpha': 0, 'beta': 1, 'weight_battery_charge': 0.02, 'weight_battery_discharge': 0.16, 'battery_minimum_state_of_charge': 0.2, 'battery_nominal_energy_capacity': 18000, 'battery_charge_power_max': 4100, 'battery_discharge_power_max': 4600, 'maximum_power_from_grid': 15000, 'maximum_power_to_grid': 5000, 'inverter_ac_output_max': 4600, 'inverter_ac_input_max': 4100, 'soc_init': 0.804, 'soc_final': 0.97, 'operating_timesteps_of_each_deferrable_load': [25], 'def_current_state': [0],`

Martijndutch, 2026-09-18:
I'm running into exactly the same issue with an EV as the deferrable load.

My requirement is slightly more specific than simply disabling the battery while the deferrable load is running.

When the EV is charging, I do **not** want the battery to discharge to supply the EV.

The desired behavior is:

* PV supplies the house first.
* The battery may discharge to cover the **house load**.
* The EV charging load must **never be supplied by the battery**.
* Any remaining EV demand should come from the grid.
* The battery should still be allowed to **charge from PV** while the EV is charging.
* When the EV is not charging, the battery should operate normally according to the EMHASS optimization.

For example:

```text
PV:          3.4 kW
House load:  1.0 kW
EV load:     7.36 kW
```

I would want EMHASS to behave approximately like this:

```text
PV -> House
PV -> EV
PV -> Battery (if there is surplus)

Battery -> House: allowed
Battery -> EV:    NOT allowed

Grid -> EV:       allowed
```

So if the house consumes 1 kW, the battery could discharge up to 1 kW to cover the house, but it should **not discharge an additional 3.5 kW (or whatever is required) to reduce the EV's grid consumption**.

In other words, when the deferrable load is active, I would like a constraint along the lines of:

```text
P_batt_discharge <= P_house_load
```

The deferrable load itself should therefore be excluded from the load that battery discharge is allowed to compensate.

This would allow EMHASS to continue optimizing the battery SOC trajectory while treating the EV as a grid-supplied deferrable load.

Currently I'm working around this in Home Assistant by putting the battery into idle mode while the EV is charging. However, this also prevents the battery from charging from PV or Grid and therefore interferes with the SOC trajectory calculated by EMHASS.

A configurable option on the deferrable load to prevent battery discharge from being used for that load would solve this much more cleanly.