davidusb-geek/emhass #558: FR: Add battery SOC sensor to EMHASS configuration

purcell-lab, 2025-06-30:
The battery SOC currently needs to be passed as a runtime parameter.

It would be useful if the battery SOC sensor could be setup with the initial EMHASS configuration and then dynamically queried by EMHASS to populate _soc_init_.

This is currently performed for _sensor_power_photovoltaics_ and _sensor_power_load_no_var_loads_

[image]

It looks like adding the relevant sensor name to the _var_list_ will pull the relevant data from Home Assistant and it is then a matter of using this result to populate _soc_init_

```
if "sensor_battery_soc" in retrieve_hass_conf:
    var_list.append(retrieve_hass_conf["sensor_battery_soc"])
```

davidusb-geek, 2025-08-26:
Hi Mark,

Could you explain how this will be useful?

I don't get this sentence:
> This is currently performed for sensor_power_photovoltaics and sensor_power_load_no_var_loads

purcell-lab, 2025-08-30:
> > This is currently performed for sensor_power_photovoltaics and sensor_power_load_no_var_loads
> 
> I probably don't get what is the request here...

This FR would make first time setups more robust.

Currently EMHASS can read PV now value and power_load now value, but cannot automatically access the battery SOC.

By being able to list the battery SOC sensor in the EMHASS config, the user doesn't need to include battery_soc as a runtime parameter, in the same fashion they don't need to include now values for PV & load.

purcell-lab, 2026-05-10:
## Status update — May 2026

**Clarifying the request** (re @davidusb-geek's question above)

The FR is to make the **initial battery SOC** discoverable from a Home Assistant sensor configured once in the EMHASS YAML, the same way `sensor_power_photovoltaics` and `sensor_power_load_no_var_loads` already are.

Today, the EMHASS config file lets the user nominate:
- a PV power sensor → EMHASS pulls the latest value from HA at runtime and uses it as the "now" point for PV
- a load power sensor → same, for net household load
- **but the battery SOC** has to be passed at every `dayahead-optim` / `naive-mpc-optim` call as the `soc_init` runtime parameter

For users with a fixed battery SOC sensor in HA (which is most of us), having to pass `soc_init` on every shell command / REST call is friction that the existing PV/load pattern has already solved. Putting `sensor_battery_soc` in the EMHASS config would mirror that pattern.

**Related issues**
- [#585](https://github.com/davidusb-geek/emhass/issues/585) — "Use Actual SOC as Starting Point in Optimization" — same underlying request, different framing.
- [#783](https://github.com/davidusb-geek/emhass/issues/783) (closed) — "soc_init not respected" — recent reliability work on the SOC pathway.

**Proposed way forward**

Minimal sketch (in `utils.py` / `retrieve_hass.py`):

```python
if "sensor_battery_soc" in retrieve_hass_conf:
    var_list.append(retrieve_hass_conf["sensor_battery_soc"])
# then after the HA pull, populate soc_init from the latest value of that sensor
# unless soc_init was explicitly passed as a runtime param (runtime wins)
```

Schema-side: add an optional `sensor_battery_soc` field to the EMHASS config schema, defaulting to unset (so existing setups are unchanged). When unset, behaviour matches today: `soc_init` must come in via runtime params.

This would also resolve [#585](https://github.com/davidusb-geek/emhass/issues/585). Happy to put up a PR if the approach sounds right.