davidusb-geek/emhass #578: on 0.13.4 open-meteo doesn't recognize the number of forecast days

g1za, 2025-08-30:
Following the update to 0.13.4 the open-meteo meteo meteo doesn't recognize the number of requested days.
I tried using both delta_forecast and delta_forecast_daily run parameters but I always get something like

[2025-08-30 12:00:42 +0200] [23] [WARNING] Open-Meteo forecast_days is too low (2) so defaulting to 3 days

Maybe something related to recent merge [538](https://github.com/davidusb-geek/emhass/pull/538) as the code was developed before the implementation of open-meteo method?

On 0.13.3 you can see below the standard output

[2025-08-30 11:55:03 +0200] [21] [INFO]  >> Obtaining params: 
[2025-08-30 11:55:03 +0200] [21] [INFO] Passed runtime parameters: {'optimization_time_step ': 30, 'delta_forecast': 2, 'prediction_horizon': 96, 'number_of_deferrable_loads': 0, 'list_nominal_power_of_deferrable_loads': [0], 'list_operating_hours_of_each_deferrable_load': [1], 'list_treat_deferrable_load_as_semi_cont': [1], 'def_current_state': [0], 'list_set_deferrable_load_single_constant': [0], 'list_start_timesteps_of_each_deferrable_load': [0], 'list_end_timesteps_of_each_deferrable_load': [0], 'soc_init': 0.97, 'soc_final': 0.2, 'battery_charge_power_max': 2500, 'battery_discharge_power_max': 7100, 'load_cost_forecast': [0.144533, 0.144533, 0.144533, 0.144533, 0.144533, 0.144533, 0.144533, 0.144533, 0.144533, 0.144533, 0.144533, 0.144533, 0.144533, 0.144533, 0.144533, 0.144533, 0.144533, 0.144533, 0.144533, 0.144533, 0.144533, 0.144533, 0.144533, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.128867, 0.144533, 0.144533, 0.12703, 0.12703, 0.12703, 0.12703, 0.12703, 0.12703, 0.12703], 'prod_price_forecast': [0.085212, 0.03682, 0.03682, 0.017884, 0.017884, 0.01052, 0.01052, 0.01052, 0.01052, 0.032612, 0.032612, 0.087895, 0.087895, 0.11572, 0.11572, 0.12624, 0.12624, 0.130942, 0.130942, 0.117824, 0.117824, 0.12098, 0.12098, 0.116109, 0.116109, 0.112059, 0.112059, 0.111985, 0.111985, 0.108356, 0.108356, 0.110565, 0.110565, 0.108756, 0.108756, 0.111544, 0.111544, 0.112564, 0.112564, 0.118045, 0.118045, 0.122032, 0.122032, 0.110565, 0.110565, 0.107136, 0.107136, 0.085212, 0.085212, 0.03682, 0.03682, 0.017884, 0.017884, 0.01052, 0.01052, 0.01052, 0.01052, 0.032612, 0.032612, 0.087895, 0.087895, 0.11572, 0.11572, 0.12624, 0.12624, 0.130942, 0.130942, 0.117824, 0.117824, 0.12098, 0.12098, 0.116109, 0.116109, 0.112059, 0.112059, 0.111985, 0.111985, 0.108356, 0.108356, 0.110565, 0.110565, 0.108756, 0.108756, 0.111544, 0.111544, 0.112564, 0.112564, 0.118045, 0.118045, 0.122032, 0.122032, 0.110565, 0.110565, 0.107136, 0.107136, 0.085212], 'alpha': 0.25, 'beta': 0.75, 'model_type': 'KNN'}
[2025-08-30 11:55:03 +0200] [21] [INFO]  >> Setting input data dict
[2025-08-30 11:55:03 +0200] [21] [INFO] Setting up needed data
[2025-08-30 11:55:03 +0200] [21] [INFO] Retrieve hass get data method initiated...
[2025-08-30 11:55:05 +0200] [21] [INFO] Retrieving weather forecast data using method = open-meteo
[2025-08-30 11:55:05 +0200] [21] [INFO] Loading existing cached Open-Meteo JSON file: /data/cached-open-meteo-forecast.json
[2025-08-30 11:55:05 +0200] [21] [INFO] The cached Open-Meteo JSON file is recent (age=0m, max_age=30m)
[2025-08-30 11:55:05 +0200] [21] [INFO] Retrieving data from hass for load forecast using method = naive
[2025-08-30 11:55:05 +0200] [21] [INFO] Retrieve hass get data method initiated...
[2025-08-30 11:55:09 +0200] [21] [INFO]  >> Performing naive MPC optimization...
[2025-08-30 11:55:09 +0200] [21] [INFO] Performing naive MPC optimization
[2025-08-30 11:55:09 +0200] [21] [INFO] Perform an iteration of a naive MPC controller
[2025-08-30 11:55:09 +0200] [21] [INFO] Status: Infeasible
[2025-08-30 11:55:09 +0200] [21] [INFO] Total value of the Cost function = 0.67
[2025-08-30 11:55:09 +0200] [21] [INFO] Optimization status: Infeasible

paulhomes, 2025-08-30:
The code I added in the PR will always fetch (and locally cache) at least 3 days of weather forecast from the Open Meteo API and so if you pass delta_forecast: 2 then you would get that warning. Reflecting on it now, a warning is not really warranted as an info or even debug would be better. Is it just the severity of the message that is an issue or is there something else going on?  I need to update to 0.13.4 tomorrow as I am still running a custom dev version in my docker container so I can get the cached weather (I fetch 7 days so I can see estimated daily PV totals for the week ahead).  I'll also work on a PR to reduce the message severity.

g1za, 2025-08-30:
I see.
Not really a problem but that warning puzzled me a bit as I could not understand if something wrong was actually happening in the background (I get more data than expected so I thought that maybe an array-size error might happen I provide emhass with less than 3 days of cost/... forecast).

If with your PR we do not just always load 3 days from open-meteo but we have the flexibility extend/shorten the number of days as much as we want (according to open-meteo API) maybe a warning/debug/... is needed only when we get less than minimum required data (maybe 1 day?).
That said, your explanation is clear and this is just related to the locally stored data, which is then used as a basis for the elaboration.

I think I could even close this ticket , as in the end it's not a bug.
Just let me know if you are ok with that.
Thanks @paulhomes !