davidusb-geek/emhass #580: using operating_timesteps_of_each_deferrable_load causes infeasible optimization results

scruysberghs, 2025-09-02:
Bug Report: operating_timesteps_of_each_deferrable_load causes infeasible optimization results

  Summary

  The runtime parameter operating_timesteps_of_each_deferrable_load causes optimization to return "infeasible" results in MPC calls, while
  the equivalent operating_hours_of_each_deferrable_load parameter works correctly with identical energy constraints.

  Environment

  - EMHASS version: [current]
  - Optimization time step: 15 minutes (4 timesteps per hour)
  - API endpoint: /action/naive-mpc-optim

  Steps to Reproduce

  Working API call (returns "Optimal"):
  {
    "operating_hours_of_each_deferrable_load": [2.0, 0.0, 0.0, 12.0],
    "prediction_horizon": 96,
    // ... other parameters
  }

  Failing API call (returns "Infeasible"):
  {
    "operating_timesteps_of_each_deferrable_load": [8.0, 0.0, 0.0, 48.0],
    "prediction_horizon": 96,
    // ... other parameters (identical to working call)
  }

  Expected Behavior

  Both calls should return "Optimal" results since they define identical energy constraints:
  - 2 hours = 8 timesteps (at 15min intervals)
  - 12 hours = 48 timesteps (at 15min intervals)

  Root Cause Analysis (done by Claude code, I don't really understand what is happening here so take it with a grain of salt)

  Located in src/emhass/optimization.py around line 870:

  ceil((60 / ((self.freq.seconds / 60) * def_total_timestep[k])) / self.timeStep)

  This time window validation formula for def_total_timestep appears to contain a mathematical error that incorrectly calculates the
  required optimization window size, making the problem infeasible.

gieljnssns, 2025-09-18:
I use "def_end_timestep".
```
    - name: Emhass mpc
      unique_id: ae62fe7e-6481-460d-a998-141e120f63ad
      state: 0
      attributes:
        mpc: >
          {
            "entity_save": true,
            "prod_price_forecast": {{
              (state_attr('sensor.ecopower_injectieprijzen', 'prijzen')[:states('sensor.prediction_horizon') | int(0)]) | tojson
              }},
            "load_cost_forecast": {{
              (state_attr('sensor.ecopower_afnameprijzen', 'prijzen')[:states('sensor.prediction_horizon') | int(0)]) | tojson
              }},
            "pv_power_forecast": {{
              ([states('sensor.gemiddelde_opbrengst_5m')|int(0)] +
              state_attr('sensor.pv_power_forecast_list', 'list')[:states('sensor.prediction_horizon') | int(0) - 1]) | tojson
              }},
            "prediction_horizon": {{ states('sensor.prediction_horizon') | int(0) }},
            "model_type": "load_forecast",
            "alpha": 1,
            "beta": 0,
            "num_def_loads": 5,
            "p_deferrable_nom": [2000, 2000, 1700, 900, 2000],
            "def_total_hours": [
              {{  (states('sensor.wasmachien_uren')|float * 4) | round(0) / 4 }},
              {{  (states('sensor.droogkast_uren')|float * 4) | round(0) / 4 }},
              {{  (states('sensor.afwasmachien_uren')|float * 4) | round(0) / 4 }},
              {{  (states('sensor.warmtepompboiler_uren')|float * 4) | round(0) / 4}},
              {{  (0 |float * 4) | round(0) / 4 }}],
            "set_def_constant":[true, true, true, true, true],
            "def_start_timestep": [
              0,
              0,
              0,
              {% if (as_timestamp(now()) - states("sensor.warmtepompboiler_is_gestopt_op")|float(0)) < 1500 -%}
                6
              {% else -%}
                0
              {%- endif -%},
              0],
            "def_end_timestep": [
              {{ [ states('sensor.wasmachien_end_timesteps') | int(0), ((states('sensor.wasmachien_uren') | float * 4) | round(0) * 3) ] | max }},
              {{ [ states('sensor.droogkast_end_timesteps') | int(0), ((states('sensor.droogkast_uren') | float * 4) | round(0) * 3) ] | max }},
              {{ [ states('sensor.afwasmachien_end_timesteps') | int(0), ((states('sensor.afwasmachien_uren') | float * 4) | round(0) * 3) ] | max }},
              {{ [ states('sensor.warmtepompboiler_end_timesteps') | int(0), ((states('sensor.warmtepompboiler_uren') | float * 4) | round(0) * 3) ] | max }},
              0],
            "def_current_state": [
              {{  "true" if is_state('input_boolean.emhass_wasmachien', "on") else "false" }}, 
              {{  "true" if is_state('input_boolean.emhass_droogkast', "on") else "false" }}, 
              {{  "true" if is_state('input_boolean.emhass_afwasmachien', "on") else "false" }}, 
              {{  "true" if is_state('binary_sensor.warmtepompboiler_aan', "on") else "false" }}, 
              {{  "true" if is_state('input_boolean.emhass_warmtepomp', "on") else "false" }}]
          }
```

davidusb-geek, 2025-09-18:
You can copy and modify the `test_perform_naive_mpc_optim` method and adapt the needed inputs to test this.
You call it something like `test_perform_naive_mpc_optim_def_total_timestep`

scruysberghs, 2025-09-18:
had claude add a unit test to the PR with your comments as input... have a look

davidusb-geek, 2025-09-18:
> had claude add a unit test to the PR with your comments as input... have a look

See my reply on the PR

davidusb-geek, 2025-12-09:
Solved at PR #583